Koenkk/zigbee2mqtt #16615: _TZE200_yojqa8xn converter broken in 1.30.1

Masterz69, 2023-02-10:
### What happened?

Device support added by https://github.com/Koenkk/zigbee2mqtt/issues/14571 .

@sercio reported issue in Z2M 1.30.1 
- https://github.com/Koenkk/zigbee2mqtt/issues/14571#issuecomment-1422426960

In my case Z2M 1.29.2 demonstrate another value in Z2M state, but same binary_sensor state: 'unknown'.

[image]

1.29.2:     "gas": "not detected",
1.30.1:     "gas": "false",

In both cases binary_sensor.xx_gas have value 'unknown'.

### Zigbee2MQTT version

1.29.2

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

Masterz69, 2023-02-11:
As well getting when try to change settings:
```
2023-02-11 14:18:14 No converter available for 'alarm_time' (121)
2023-02-11 14:19:31 No converter available for 'alarm_ringtone' ("2")
```

Koenkk, 2023-02-12:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_yojqa8xn']),
    model: 'TS0601_gas_sensor_2',
    vendor: 'TuYa',
    description: 'Gas sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        exposes.enum('alarm_ringtone', ea.STATE_SET, ['1', '2', '3', '4', '5']).withDescription('Ringtone of the alarm'),
        exposes.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        exposes.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas',
                {
                    from: (v) => {
                        return v === tuya.enum(0);
                    },
                },
            ],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({'1': 0, '2': 1, '3': 2, '4': 3, '5': 4})],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Masterz69, 2023-02-12:
> * start z2m, check if issue is fixed

Installed 1.30.1.-1., ext.converter added, Z2M restarted, sensor forced to send data by pressing button.

```binary_sensor``` looks fine now, but still no converter trying to change ```alarm_ringtone``` & ```alarm_time```.

[image]

```
error 2023-02-12 11:29:01: No converter available for 'alarm_ringtone' ("3")
error 2023-02-12 11:29:02: No converter available for 'alarm_time' (31)
```

Masterz69, 2023-02-12:
When button pressed on device.

MQTT Explorer show topic value:
```
{
  "alarm_ringtone": "RingTone 2",
  "alarm_switch": false,
  "alarm_time": 120,
  "gas": false,
  "gas_value": 0,
  "last_seen": "2023-02-12T12:07:41+02:00",
  "linkquality": 84,
  "preheat": false,
  "self_test": false,
  "self_test_result": "success",
  "silence": null
}
```

HA log:
```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:154
Integration: MQTT (documentation, issues)
First occurred: 25 January 2023 at 14:55:50 (132203 occurrences)
Last logged: 12:07:41

Invalid option for select.sgas_boiler_alarm_ringtone: 'RingTone 2' (valid options: ['1', '2', '3', '4', '5'])
```

Koenkk, 2023-02-13:
I pushed some fixes, can you check the latest dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Masterz69, 2023-02-21:
As @sercio confirmed in https://github.com/Koenkk/zigbee2mqtt/issues/14571#issuecomment-1430892834:
"
OK, tested dev brach and for me all is solved:

Correct Value on gas attribute, no error when try to set Alarm Ringtone and Alarm Time.
"

Koenkk, 2023-02-22:
Great!

closed 2023-02-22 by Koenkk

Roman-RV78, 2023-07-13:
Hello. There is a problem with gas detected, the value is always false. Can anyone confirm this with this device? As I understand when pressing self_test, gas detect should be true