Koenkk/zigbee2mqtt #16618: Sengled E21-N1EA OTA fails with error "No image available for imageType '32972'"

alexwaibel, 2023-02-11:
### What happened?

When I attempt to check for an OTA for the [Sengled E21-N1EA](https://www.zigbee2mqtt.io/devices/E21-N1EA.html#sengled-e21-n1ea) I see an error pop up in the top right that says "No image available for imageType '32972'". I have 13 of these bulbs and have seen this same behavior on all of them. I'm using a Zooz ZST10 700 but have seen this same error on my HUSBZB-1.

Digging through the [OTA json](https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json) I see several Sengled devices but none that match this image type "32972" or contain "E21-N1EA". Based on that I believe there's not yet an OTA image available for these bulbs in z2m, in which case I'd expect them not to show up in the OTA tab. However, I also see that these are tagged in the docs as OTA compatible and I also see reports on this issue of some users being able to OTA the bulbs:
https://github.com/Koenkk/zigbee2mqtt/issues/12189

### What did you expect to happen?

If there is an image for these bulbs, I'd expect a different error message (maybe I'm already on latest FW version?). Otherwise, I find it confusing that these bulbs show up in the OTA tab despite not having an entry in the OTA json.

I see a few other similar GH issues for this from late last year (mostly auto marked stale) so seems others are confused by this:
https://github.com/Koenkk/zigbee2mqtt/issues?q=E21-N1EA

### How to reproduce it (minimal and precise)

1. Join Sengled E21-N1EA bulb to z2m network
2. Navigate to OTA tab
3. Click "Check for new updates"

### Zigbee2MQTT version

1.30.0 commit: 8f781db

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Zooz ZST10 700

### Debug log

```
info 2023-02-10 16:34:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Living Room Lamp'","meta":{"device":"Living Room Lamp","status":"checking_if_available"},"type":"ota_update"}'
info 2023-02-10 16:34:38MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Living Room Lamp' (No image available for imageType '32972')","meta":{"device":"Living Room Lamp","status":"check_failed"},"type":"ota_update"}'
info 2023-02-10 16:34:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Living Room Lamp"},"error":"Failed to check if update available for 'Living Room Lamp' (No image available for imageType '32972')","status":"error","transaction":"ldr11-20"}'
```

Koenkk, 2023-02-12:
What message would you expect?

alexwaibel, 2023-02-17:
"No image available" does accurately capture the issue, maybe [the troubleshooting section of the OTA wiki page](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#troubleshooting) could mention this though? I just found it initially unclear that although these devices are called out in [their wiki page](https://www.zigbee2mqtt.io/devices/E21-N1EA.html#sengled-e21-n1ea) as supporting OTAs and show up in the z2m OTA tab, there may not be any images published.

closed 2023-03-27 by github-actions[bot]

aryanakh, 2023-04-30:
Still a major issue for me too. Did anyone find a solution? I have issues with my bulbs dropping from my network

Stefkeys, 2023-10-21:
I have the same issue. Can this issue be reopened?

rtpguy, 2024-01-03:
Looks like this is still an issue as of 1.35.0.  I just moved from dev to release hoping it was resolved after seeing this ticket but looks like it's still a problem.
Error 2024-01-03 15:29:41Failed to check if update available for 'Office Sengled' (No image available for imageType '32972')

jroger0707, 2024-01-12:
I am on 1.35.1 and see the same issue on the zigbee2mqtt side.  I have 18 of these bulbs.  Further, I am using Home Assistant and am getting prompted about firmware updates; current version noted is 38, latest version noted is -1.  I select to ignore this update but it keeps coming back for all 18 bulbs.  Something is not working correctly.  If there is no firmware update available then zigbee2mqtt is not handling something correctly.

nberardi, 2024-02-11:
This issue was marked as closed. Devs do we need to reopen it?

MRobi1, 2024-02-11:
Still an issue for me so not sure if I missed the solution to warrant this being closed