Koenkk/zigbee2mqtt #16628: Warning installing Aqara FP1. Unknown key 0x010c = 3"

misterwobbley, 2023-02-11:
### What happened?

Whene I pair up my Aqara FP1 Pesence sensor I get the warning 

"warning 2023-02-11 20:41:03zigbee-herdsman-converters:xiaomi:aqara_fp1: Unknown key 0x010c = 3"

This occurs both in the regular branch and in the Edge package.

I am running on the HA OS on a raspberry pi, Home Assistant 2023.2.3, Supervisor 2023.01.1, Operating System 9.5 and Frontend 20230202.0

My Sonoff Zigbee Plus dongle has been upgraded to 20221226, and the firmware of the FP1 sensor is 54

### Zigbee2MQTT version

1.30.1 and 1.30.1-dev commit: e9a3835

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10714752/log.txt)

11harveyj, 2023-02-13:
Same problem here, with pretty much a identical setup (Sonoff ZB plus USB dongle, firmware 20221226).

saschachina, 2023-02-13:
I use the Conbee2 with the version 0x26780700, MQTT version 1.30.1

11harveyj, 2023-02-13:
@saschachina try re-pairing again, it seems to just be a warning more than anything as it hasn't actually impacted it's functionality (apart from on first pair where it was online, but nothing was updating)

saschachina, 2023-02-14:
@11harveyj I have tried three times to pair the FP1, first I deleted the device from Zigbeemqtt, then reset the device by pressing the button for about 10 seconds and then repaired.
Unfortunately all without success, the FP1 shows me constantly presence and only changes to absent when I press reset_nopresence_status.
If I go into the room then a few seconds later the presence will be showed but doesn´t clear it.

lux73, 2023-03-04:
just to be clear, you have to set a Rule that trigger "reset_nopresence_status" in your preferred HomeAutomation Evironment that resets the "presence" State to false after the "presence_event" detects "away" or "leave" the Room

works for me without any hassle since four Weeks & i think so it's designed

arteck, 2023-03-04:
same problem here.. [i found in converter a async function which ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/348dc94d3daaba5d7041410e1dda274af236b23f/lib/xiaomi.js#L146) this function with async operation causes problems.. 
i think the switch to async is causing problems

saschachina, 2023-03-04:
> same problem here.. [i found in converter a async function which ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/348dc94d3daaba5d7041410e1dda274af236b23f/lib/xiaomi.js#L146) this function with async operation causes problems.. this came in version .37 from converter and from then on there were problems i think the switch to async is causing problems

This is exactly the problem I have, no idea what to do here. 
I already have a FP1 in the bathroom and it works perfectly for months without problems.

Here a comparison of the two devices:

The Problem maker:
`{"id":37,"type":"EndDevice","ieeeAddr":"0x54ef44100075fb36","nwkAddr":58695,"manufId":4660,"manufName":"aqara","powerSource":"DC Source","modelId":"lumi.motion.ac01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":65520,"inClusterList":[0,3,64704],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.motion.ac01","appVersion":54,"swBuildId":""}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,29,5,33,1,0,8,33,54,1,10,33,144,142,12,32,20,16,32,1,18,32,0,101,32,1,102,32,3,103,32,0,104,32,0,105,32,1,106,32,1,107,32,3]},"268":3,"322":1,"323":7,"324":0,"326":1,"336":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":54,"stackVersion":2,"hwVersion":1,"dateCode":"20210121","swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2044002991},"lastSeen":1677957672835,"defaultSendRequestWhen":"immediate"}`

The device it works perfect in the bathroom:
`{"id":20,"type":"EndDevice","ieeeAddr":"0x54ef44100053c348","nwkAddr":21962,"manufId":4660,"manufName":"aqara","powerSource":"DC Source","modelId":"lumi.motion.ac01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":65520,"inClusterList":[0,3,64704],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.motion.ac01","manufacturerName":"aqara","powerSource":4,"zclVersion":3,"appVersion":54,"stackVersion":2,"hwVersion":1,"dateCode":"20210121","swBuildId":""}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,22,5,33,4,0,8,33,54,1,10,33,121,176,12,32,20,16,32,1,18,32,0,101,32,0,102,32,3,103,32,0,104,32,0,105,32,1,106,32,1,107,32,3]},"268":3,"322":0,"323":1,"324":0,"326":1,"336":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":54,"stackVersion":2,"hwVersion":1,"dateCode":"20210121","swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2044002991},"lastSeen":1677957644821,"defaultSendRequestWhen":"immediate"}`

saschachina, 2023-03-04:
> just to be clear, you have to set a Rule that trigger "reset_nopresence_status" in your preferred HomeAutomation Evironment that resets the "presence" State to false after the "presence_event" [...]

Thanks a lot, it could work but cant be the final solution.

11harveyj, 2023-03-04:
> just to be clear, you have to set a Rule that trigger "reset_nopresence_status" in your preferred HomeAutomation Evironment that resets the "presence" State to false after the "presence_event" detects "away" or "leave" the Room

Not something I need to do here, its presence is updating automatically for me now with no issues. Generally takes a couple of seconds to update.

saschachina, 2023-03-04:
> Not something I need to do here, its presence is updating automatically for me now with no issues. Generally takes a couple of seconds to update.

Thats what the FP1 normaly do or what he have to do.

t34wrj, 2023-03-12:
I am also having this issue.

saschachina, 2023-03-13:
Last Saturday I changed my system completely from MQTT to ZHA because of other reasons, the FP1 still occupancy will not clear! So it have nothing do with ZHA or ZigbeeMQTT!
I changed the position of the FP1 in any possible environment, changed the room, tried different power supplies!
Nothing helps and i get annoyed, maybe hardware problem?

t34wrj, 2023-03-15:
> just to be clear, you have to set a Rule that trigger "reset_nopresence_status" in your preferred HomeAutomation Evironment that resets the "presence" State to false after the "presence_event" [...]

I was having a similar problem with the room presence not clearing and set-up and automation to do this which worked to an extent. However, after stopping this automation, I discovered that the reason that presence_events were not clearing was due to the sensitivity/approach distance settings and the placement of the sensor. Once I tweaked those to my room, I no longer needed the reset_nopresence automation.

Caros2017, 2023-05-11:
Did anyone know how to solve the `Unknown key 0x010c = 3 issue`? I am facing the exact same.

Caros2017, 2023-05-11:
> My FP1 was working fine until today, when I now see the error in Z2M, and the presence does not flip to false when away...
> 
> Also presence_event does not change from approach after leaving the room, so can't reliably use that as an automation
> 
> If I do a reset_no_presence call, event does update to leave and presence False, and does work correctly when I enter the room, but then when I exit it doesn't reset to unoccupied without me manually resetting

I have the exact same behavior on my new (first) Aqara fp1. I made a flow that does `reset_no_presence` all the time and hence I have opened a new issue #17643, since this one is closed. Can you share your findings there?

closed 2023-05-12 by Koenkk