Koenkk/zigbee2mqtt #16630: [New device support]:  SkydanceWZ5 RGBCCT  TS0505B _TZB210_3zfp8mki

BandBxx, 2023-02-12:
### Link

https://www.aliexpress.com/item/1005003145273010.html?spm=a2g0o.store_pc_groupList.8148356.7.7ef634dacMwGdo&pdp_npi=2%40dis%21RUB%212%C2%A0463%2C46+%D1%80%D1%83%D0%B1.%211%C2%A0600%2C96+%D1%80%D1%83%D0%B1.%21%21%21%21%21%4021038ed816761598093964072e1431%2112000024344552632%21sh

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1386daef7f305","nwkAddr":55019,"manufId":4417,"manufName":"_TZB210_3zfp8mki","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1676158819559,"defaultSendRequestWhen":"immediate"}

### Comments

Hello. I try to connect new SkyDance WZ5  zigbee RGBCCT  dimmer,   but it didn't  identified correctly.
Please, add it in Edge version or  help me pls  with external converter. 
Thx.

**Log in Zigbee2MQTT:**
Info 2023-02-12 03:03:50MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"RGBW_WZ5","ieee_address":"0xa4c1386daef7f305"},"type":"device_leave"}'
Info 2023-02-12 03:03:50MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1386daef7f305"},"type":"device_removed"}'
Debug 2023-02-12 03:03:51Received Zigbee message from 'TempHumHall', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":980,"scale":-1,"scaledValue":9800}' from endpoint 1 with groupID 0
Info 2023-02-12 03:03:51MQTT publish: topic 'zigbee2mqtt/TempHumHall', payload '{"battery":100,"elapsed":50443,"humidity":37.26,"last_seen":"2023-02-12T03:03:52+03:00","linkquality":144,"power_outage_count":9,"pressure":980,"temperature":23.34,"voltage":3015}'
Info 2023-02-12 03:03:54Device 'RGBW_WZ5' joined
Info 2023-02-12 03:03:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"RGBW_WZ5","ieee_address":"0xa4c1386daef7f305"},"type":"device_joined"}'
Info 2023-02-12 03:03:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"RGBW_WZ5"},"type":"device_connected"}'
Info 2023-02-12 03:03:54MQTT publish: topic 'zigbee2mqtt/RGBW_WZ5', payload '{"last_seen":"2023-02-12T03:03:55+03:00","linkquality":176}'
Info 2023-02-12 03:03:54Starting interview of 'RGBW_WZ5'
Info 2023-02-12 03:03:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"RGBW_WZ5","ieee_address":"0xa4c1386daef7f305","status":"started"},"type":"device_interview"}'
Info 2023-02-12 03:03:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"RGBW_WZ5"},"type":"pairing"}'
Debug 2023-02-12 03:03:55Received Zigbee message from 'RGBW_WZ5', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug 2023-02-12 03:03:55Skipping message, definition is undefined and still interviewing
Info 2023-02-12 03:03:55MQTT publish: topic 'zigbee2mqtt/RGBW_WZ5', payload '{"last_seen":"2023-02-12T03:03:55+03:00","linkquality":184}'
Debug 2023-02-12 03:03:55Received Zigbee message from 'RGBW_WZ5', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug 2023-02-12 03:03:55Skipping message, definition is undefined and still interviewing
Info 2023-02-12 03:03:55MQTT publish: topic 'zigbee2mqtt/RGBW_WZ5', payload '{"last_seen":"2023-02-12T03:03:55+03:00","linkquality":180}'
Debug 2023-02-12 03:03:55Received Zigbee message from 'RGBW_WZ5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2023-02-12 03:03:55Skipping message, definition is undefined and still interviewing
Info 2023-02-12 03:03:55MQTT publish: topic 'zigbee2mqtt/RGBW_WZ5', payload '{"last_seen":"2023-02-12T03:03:55+03:00","linkquality":176}'
Info 2023-02-12 03:03:55Successfully interviewed 'RGBW_WZ5', device has successfully been paired
Warning 2023-02-12 03:03:55Device 'RGBW_WZ5' with Zigbee model 'TS0505B' and manufacturer name '_TZB210_3zfp8mki' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-02-12 03:03:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"RGBW_WZ5","ieee_address":"0xa4c1386daef7f305","status":"successful","supported":false},"type":"device_interview"}'
Info 2023-02-12 03:03:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"RGBW_WZ5","supported":false},"type":"pairing"}'

### Supported color modes

RGBCCT

BandBxx, 2023-02-15:
I tried use this external converters: https://githubplus.com/Koenkk/zigbee2mqtt/issues/13446
But  first one is incorrect with  white light  (constantly ON) and the second have not any effect on any buttons, didnt work

BandBxx, 2023-02-19:
The code in https://github.com/Koenkk/zigbee2mqtt/issues/13466  is works,  but  no white channel  and  the brightness worked wrong:   any changement lead to white color

jonwilliams84, 2023-03-20:
Just picked up some of these to replace the WT5 (WiFi versions) as they randomly stop working with Tuya local occasionally.

They have paired to Z2M fine but the colours are not correct.  

If they are controlled via an external remote control the colours are spot on so it appears to be a mapping issue in Z2M. 

Also, is there anyway to keep the state of the controller in sync with Z2M as when controlled via external remote the state needs to be manually refreshed in Z2M for it to update?

Thanks

Jon

jonwilliams84, 2023-03-20:
> Just picked up some of these to replace the WT5 (WiFi versions) as they randomly stop working with Tuya local occasionally.
>
> They have paired to Z2M fine but the colours are not correct.
>
> If they [...]

My device is actually this (or a derivative of this): https://www.zigbee2mqtt.io/devices/WZ5_rgbcct.html

However it's being detected as this:

{"id":37,"type":"Router","ieeeAddr":"0xa4c1389c0eadc306","nwkAddr":49374,"manufId":4417,"manufName":"_TZB210_3zfp8mki","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZB210_3zfp8mki","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":27,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":45494,"currentY":19418,"currentHue":16,"currentSaturation":16,"colorTemperature":153}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1679354028239,"defaultSendRequestWhen":"immediate"}

I believe the problem may be with the colour modes supported is per the skydance wz5 rgbcct product and not the color_xy as per the TS0505B device that Z2M believes it to be.

Can I override this somehow and get Z2M to detect the device as the Skydance wz5 rgbcct?

BandBxx, 2023-03-21:
Yes, good idea!  in the last few releases this device has been defined in Z2M, but like "as TUYA TS_XXXX", not Skydance. The problem with adjusting the brightness of colors and constant white light is not solved (

closed 2023-04-29 by github-actions[bot]

LeoS3, 2023-10-06:
Any updates on this issue? I have wzs1 and wzs5 identified as Tuya TS0505B like above

bzumik1, 2023-11-14:
I have WZS2, that is CCT version. Is there any progress on this?

LaurentiuGrama, 2024-01-28:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

I have the same issue. 
Unfortunately, it seems that we have no fix for this problem