Koenkk/zigbee2mqtt #16637: Remote Control N2 E2001/E2002  shows 180% battery level

aolik08, 2023-02-12:
### What happened?

I have a Ikea Styrbar Remote Control N2. After updating the remote to the latest version 2.4.5, it shows the battery levels above 100%, more precisely: 180%. It's obviously due to the latest firmware update. This is not a critical issue, but thought I'd mention it.

[image]

### What did you expect to happen?

Battery level should not exceed 100%.

### Zigbee2MQTT version

1.30.1-dev

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

aolik08, 2023-02-12:
```
{
    "battery": 180,
    "last_seen": "2023-02-12T10:09:52+02:00",
    "linkquality": 72,
    "update": {
        "installed_version": 33816581,
        "latest_version": 581,
        "state": "idle"
    }
}
```

Koenkk, 2023-02-13:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5465 by @sjorge 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-13 by Koenkk

aolik08, 2023-02-14:
Thank you @Koenkk and @sjorge 

In this case, I believe taht these lines can be safely removed/altered:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/505dc9809be48b493024e639aa498204cb01873d/devices/ikea.js#L671 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/505dc9809be48b493024e639aa498204cb01873d/devices/ikea.js#L693 (only ', battery: {dontDividePercentage: true}' part)
https://github.com/Koenkk/zigbee-herdsman-converters/blob/505dc9809be48b493024e639aa498204cb01873d/devices/ikea.js#L705 (only ', battery: {dontDividePercentage: true}' part)
https://github.com/Koenkk/zigbee-herdsman-converters/blob/505dc9809be48b493024e639aa498204cb01873d/devices/ikea.js#L732 (only ', battery: {dontDividePercentage: true}' part)
https://github.com/Koenkk/zigbee-herdsman-converters/blob/505dc9809be48b493024e639aa498204cb01873d/devices/ikea.js#L871

sjorge, 2023-02-14:
Correct, I missed those. They can indeed be removed.