Koenkk/zigbee2mqtt #16641: [Feature request]: Zibee Direct Support

danieledwardgeorgehitchcock, 2023-02-12:
### Is your feature request related to a problem? Please describe

There is a new addition to the Zigbee standard - Zigbee Direct allows
for encrypted and authenticated access to Zigbee devices and networks from
Bluetooth Low Energy ("BLE") capable devices. This enables devices like
smartphones, tablets, smart speakers and hubs to commission and control
devices on a Zigbee network.

### Describe the solution you'd like

Investigation of the feasibility of adding Zigbee Direct to the Z2M codebase, and possible implementation

### Describe alternatives you've considered

None - this is just bringing the subject into light

### Additional context

Here is the spec: [Zigbee Direct Specification](https://csa-iot.org/wp-content/uploads/2022/12/20-27688-037-zigbee_direct_spec.pdf)

Danit2, 2023-02-12:
👍

Koenkk, 2023-02-13:
I saw about this, but this doesn't seem to be applicable for z2m. The use case is using e.g. your phone/tablet to directly control Zigbee devices over bluetooth without the need of a gateway (and not Zigbee). Therefore I think you can already use Zigbee direct in a Zigbee2MQTT Zigbee network.

danieledwardgeorgehitchcock, 2023-02-13:
It is aimed at hubs too but agree it is mostly geared around BLE control of a Zigbee network via a compatible device.

The way I saw it integrating with Z2M is as an extension (maybe run as a separate process?)

An interesting use case (and how I'd use it) is in a High Availability environment where you could have several of these BLE control devices able to manage your network irrespective of which sever you're running on - something desperately missing from the Zigbee specification with its single coordinator

Koenkk, 2023-02-13:
@danieledwardgeorgehitchcock I don't think this should be part of the z2m code base, it sounds more like a separate project. This also will require some heavy modifications to the firmware to support both bluetooth and zigbee at the same time.

closed 2023-02-13 by danieledwardgeorgehitchcock

ciB89, 2023-02-18:
> I saw about this, but this doesn't seem to be applicable for z2m. The use case is using e.g. your phone/tablet to directly control Zigbee devices over bluetooth without the need of a gateway (and not Zigbee). Therefore I think you can already use Zigbee direct in a Zigbee2MQTT Zigbee network.

As far as I understood it, it is to bypass the hub/coordinator, but not only to control the light that has BLE, but any other zigbee device connected to the network in general. The BLE device would act as a kind of "border router".