Koenkk/zigbee2mqtt #16643: Device 'TS0601', manufacturerName: '_TZE200_gd4rvykv' recognized as a wrong model

pioniziolek, 2023-02-12:
### What happened?

modelID: 'TS0601', manufacturerName: '_TZE200_gd4rvykv' wrongly recognized as TS0601_thermostat.

### What did you expect to happen?

This device looks exactly like TS0601_thermostat3 but not TS0601_thermostat. Possibly it is was assigned to wrong model.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10716629/log.txt)

pioniziolek, 2023-02-12:
Please discard previous log. Use following:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10716724/log.txt)
Zigbee Manufacturer reported: _TZE200_gd4rvykv
IEEE Address: 0xa4c138b76d087085

Just by picture evaluation device is different than 'TS0601_thermostat' but looks exactly like 'TS0601_thermostat'.

Koenkk, 2023-02-13:
See https://github.com/Koenkk/zigbee2mqtt/discussions/16642

closed 2023-02-13 by Koenkk