Koenkk/zigbee2mqtt #16645: Device Niko 552-72301 state reported wron

MG2R, 2023-02-12:
### What happened?

When the Zigbee Motor Control switch from Niko (Niko 552-72301) has its currentLiftPercentage at 0, the switch thinks the blinds are in the closed position and lights up only the "open button". The "state" according to zigbee2mqtt is set to OPEN. 

Similarly, when the currentLiftPercentage is 100, the switch assumes the blinds it's controlling are fully open and only lights up the "close button". The "state" according to zigbee2mqtt is set to "CLOSE".

This causes the icons and the buttons in e.g. Home Assistant to be wrong and requires templates to inverse the state.

Switching the "invert_cover" allows me to fix the buttons. However, the icons track the state field and that is not changed by the "invert_cover" switch.

It seems like the state is tracking the LEDs, which is inverse because the LED to close the blinds lights up when the blinds are open.

### What did you expect to happen?

When the blinds are closed, the state should be "CLOSE", when the blinds are open the state should be "OPEN".

### How to reproduce it (minimal and precise)

Hook the switch up and either open or close the covers.

### Zigbee2MQTT version

1.30.1 as well as latest-dev (docker)

### Adapter firmware version

6.10.3.0 build 297

### Adapter

QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2

MG2R, 2023-02-12:
Note: support for these was added in #12755

@bjorndeboodt, you confirmed they were working properly. Could you also have a look at the state and see if you can confirm my findings?

MG2R, 2023-03-19:
Still relevant to me

closed 2023-04-27 by github-actions[bot]

nikplas, 2024-08-24:
Same probleem here, but i use ZHA