Koenkk/zigbee2mqtt #16646: Third Reality smart plug with power missing decimal

artbird309, 2023-02-13:
### What happened?

I added four new ZigBee smart plugs Third Reality BLE smart plug with power (3RSP02028BZ) it connects fine and turns off and on but the power metering data is without a decimal point (e.g. 1159 instead of 115.9). This plug was recently added by @weihuan1111 in PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/5093

State: from one of the plugs: 
`{
    "current": 26,
    "last_seen": "2023-02-12T19:15:28-05:00",
    "linkquality": 109,
    "power": 13,
    "state": "ON",
    "voltage": 1206
}`

### What did you expect to happen?

I expected the state to be 1.3W and 120.6V.

### How to reproduce it (minimal and precise)

Connect 3RSP02028BZ and load and read power metering data. 

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

 20221226

### Adapter

SMLIGHT SLZB-06 running CC2652P

Vendo232, 2023-02-24:
I have the same problem

[image]

Vendo232, 2023-02-24:
using custom converter does not help either

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

module.exports = [
    {
        zigbeeModel: ['3RSP02028BZ'],
        model: '3RSP02028BZ',
        vendor: 'Third Reality',
        description: 'Zigbee / BLE smart plug with power',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        ota: ota.zigbeeOTA,
        exposes: [e.switch(), e.ac_frequency(), e.power(),e.power_factor(), e.energy(), e.current(), e.voltage().withAccess(ea.STATE)],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await reporting.onOff(endpoint);
            await reporting.activePower(endpoint, {change: 10});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.readMeteringMultiplierDivisor(endpoint);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 10, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1, acPowerDivisor: 10,
            });
            device.save();
        },
    },
];`

[image]

Automayted, 2023-02-26:
Same issue here.

Koenkk, 2023-02-26:
Can you provide the debug log when the device reports power such that I can check if decimals are reported?

Vendo232, 2023-02-26:
Hope thia is it

[image]

artbird309, 2023-02-26:
Below is the debug log for one of my plugs, I do not see decimals being reported for any of the measurements (current, power, or voltage).
`Debug 2023-02-26 15:08:41Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":15}' from endpoint 1 with groupID 0
Info 2023-02-26 15:08:41MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":15,"last_seen":"2023-02-26T15:08:40-05:00","linkquality":91,"power":12,"state":"ON","voltage":1212}'
Debug 2023-02-26 15:08:41Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":15}' from endpoint 1 with groupID 0
Info 2023-02-26 15:08:41MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":15,"last_seen":"2023-02-26T15:08:40-05:00","linkquality":94,"power":12,"state":"ON","voltage":1212}'
Debug 2023-02-26 15:08:42Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0
Info 2023-02-26 15:08:42MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":15,"last_seen":"2023-02-26T15:08:41-05:00","linkquality":94,"power":7,"state":"ON","voltage":1212}'
Debug 2023-02-26 15:08:42Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0
Info 2023-02-26 15:08:42MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":15,"last_seen":"2023-02-26T15:08:41-05:00","linkquality":65,"power":7,"state":"ON","voltage":1212}'
Debug 2023-02-26 15:12:41Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":16}' from endpoint 1 with groupID 0
Info 2023-02-26 15:12:41MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":16,"last_seen":"2023-02-26T15:12:40-05:00","linkquality":105,"power":7,"state":"ON","voltage":1212}'
Debug 2023-02-26 15:12:41Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":1214}' from endpoint 1 with groupID 0
Info 2023-02-26 15:12:41MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":16,"last_seen":"2023-02-26T15:12:40-05:00","linkquality":87,"power":7,"state":"ON","voltage":1214}'
Debug 2023-02-26 15:12:42Received Zigbee message from 'plug_name', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0
Info 2023-02-26 15:12:42MQTT publish: topic 'zigbee2mqtt/plug_name', payload '{"current":16,"last_seen":"2023-02-26T15:12:41-05:00","linkquality":112,"power":7,"state":"ON","voltage":1214}'`

weihuan1111, 2023-02-27:
I use our latest version(v1.00.41) of plug to test it. 
I put this code to my Z2M to test localy,  [https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/third_reality.js]
And this issue has been fixed, you can see screenshot below.
I will submit ota file to zigbee2mqtt, so you can update your plug to the latest.
[image]

Vendo232, 2023-02-27:
is use of converter file always going to be required? 

I`m on latest-dev released today and FW 1.00.41 and I do not see difference but I`m not using the js converter though.

weihuan1111, 2023-02-28:
> it is fixed now, updated to 1.00.41 and restarted the z2m not using external converters
> 
> thank you.
> 
> [image]

We have fix the energy issue in zigbee2mqtt and you can wait for the next version of zigbee2mqtt.

Koenkk, 2023-02-28:
Fix will be included in the 1 February release. Thanks @weihuan1111 !

closed 2023-02-28 by Koenkk

Automayted, 2023-02-28:
I can confirm the latest-dev container and device firmware v1.00.41 resolved the voltage, current, and power readings, but total energy (kWh) is still not correct.

It appears the total energy reading is off by 100,000x - my 8.4W test load running for an hour reports 894.3kWh, not .008943kWh.

`{
    "ac_frequency": 60,
    "current": 0.13,
    "energy": 894.3,
    "last_seen": "2023-02-28T12:16:53-06:00",
    "linkquality": 108,
    "power": 8.4,
    "state": "ON",
    "update": {
        "installed_version": 268513322,
        "latest_version": 268513322,
        "state": "idle"
    },
    "voltage": 122,
    "power_factor": null,
    "update_available": null
}`

tgkenney, 2023-03-29:
I have zigbee2mqtt version 1.30.2 running as a HAOS add-on, and firmware version v1.00.41 running on the device, but I'm still seeing this issue I believe.

I'm also not getting any energy reading from the device, the field is just showing as null.  No clue if I'm doing something wrong there.

`{
    "ac_frequency": 60,
    "current": 266,
    "linkquality": 212,
    "power": 30.2,
    "state": "ON",
    "update": {
        "installed_version": 268513322,
        "latest_version": 268513322,
        "state": "idle"
    },
    "voltage": 1185,
    "energy": null,
    "power_factor": null,
    "update_available": null
}`