Koenkk/zigbee2mqtt #16654: CC2652RB (slae.sh) Routers Offline with CC2652RB_router_20221102.hex firmware

thetravellor, 2023-02-13:
### What happened?

The two CC2652RB sticks can be flashed successfully using cc2652bsl, they pair with zigbee2mqtt after pairing.

They may (or may not) appear online briefly in zigbee2mqtt, but soon go offline 

They may be switched to "online" using the reconfigure device button, however when they are pinged by zigbee2mqtt they immediately go offline.

The devices were working earlier on the old fimware (CC2652RB_router_20220125), however I cannot download https://github.com/Koenkk/Z-Stack-firmware/blob/Z-Stack_Home_3.x.0_router_20220125/router/Z-Stack_3.x.0/bin/CC2652RB_router_20220125.zip as I just get a corrupted zip file that cannot be opened with unzip, so I cannot downgrade also.

Its happening with two of these sticks.

### What did you expect to happen?

I expected the sticks to pair successfully (I have two) and work as they had previously.

### How to reproduce it (minimal and precise)

Flash slae.sh 2652Rb stcik with this version of the router firmware

### Zigbee2MQTT version

1.30.1 commit: eb878d3d

### Adapter firmware version

20221226

### Adapter

Slae.sh 2652RB stick

### Debug log

[picture.pdf](https://github.com/Koenkk/zigbee2mqtt/files/10721488/picture.pdf)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10721489/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10721516/log.txt)
[image]

jkempson, 2023-02-25:
I'm having the exact same issue

thetravellor, 2023-03-28:
not stale, still an issue

thetravellor, 2023-04-29:
@Koenkk please look at this mate, or at least comment?

Koenkk, 2023-04-29:
I'll compile a new firmware once the coordinator once is ready (https://github.com/Koenkk/Z-Stack-firmware/issues/448). Until then, please keep using the old firmware.

closed 2023-06-06 by github-actions[bot]

ssm84, 2026-02-28:
i still have the same isue using latest router build 20250403