Koenkk/zigbee2mqtt #16655: [Feature request]: Danalock OTA-Updates

Schluggi, 2023-02-13:
### Is your feature request related to a problem? Please describe

I was recently in contact with the Danalock-support and asked for publicly available firmware files for OTA-Updates. 
Today a got a mail: 
"I promised to let you know when the Zigbee firmware update files and program were online. [...]"

https://danalock.com/zigbee-firmware-update

### Describe the solution you'd like

Providing OTA-Updates for the Danalock V3.

### Describe alternatives you've considered

-

### Additional context

-

Schluggi, 2023-02-13:
I just found out, how OTA updates in zigbee2mqtt work and created a PR: https://github.com/Koenkk/zigbee-OTA/pull/258

Schluggi, 2023-02-13:
@Koenkk 
After you merged https://github.com/Koenkk/zigbee-OTA/pull/258 is there anything else to do? How can I help?

My Danalock is not shown up at the OTA menu yet.

Koenkk, 2023-02-13:
I've enabled the OTA, it will be available in the next release (1 march) and the dev branch in a few hours from now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-13 by Koenkk