Koenkk/zigbee2mqtt #16657: IKEA E1743 TRADFRI ON/OFF switch leaves network after switch from CC2531 to SkyConnect

Desttro, 2023-02-13:
### What happened?

Hello. I was using **CC2531** adapter before I switched to **SkyConnect** from Home Assistant. All devices work fine except the **IKEA E1743 TRADFRI ON/OFF switch**.

These devices regularly leave the network. As a workaround, I am now running two instances of zigbee2mqtt, each with a different adapter and channel set. It works fine, but it's not an optimal solution because they have to interfere with each other.

I have updated everything to the latest stable firmware, but the problem still persists.

- Zigbee2MQTT: **1.30.1 commit: eb878d3**
- Coordinator: **EZSP v9, 7.1.4.0 build 389**
- TRÅDFRI on/off switch: **2.3.079 / {"installed_version":587699761,"latest_version":587699761}**

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

7.1.4.0 build 389

### Adapter

Home Assistant SkyConnect

### Debug log

[debug.log](https://github.com/Koenkk/zigbee2mqtt/files/10721772/debug.log)

Wireheadbe, 2023-03-07:
Testing with:
- E1743: 2.2.010
- Z2M: 1.30.2
- Coordinator: EZSP v10, 7.2.1.0 build 144

Will see if I get any similar behaviour. Might be worth updating your firmware to the latest version.

Desttro, 2023-03-08:
I have updated the coordinator to version 7.2.1.0 build 144. Issue persists.

Wireheadbe, 2023-03-08:
Are you sure the interview finished completely? Can you try to click reconfigure while waking up the device?
No outage after 24 hours here with versions mentioned above.

Wireheadbe, 2023-03-12:
Still no drops here. Battery usage normal as well. Version 1.30.2, commit cdf62ea / EZSP v10, 7.2.1.0 build 144 / E1743v2.2.010

Desttro, 2023-03-13:
Yes, the interview finished successfully. I also tried reconfiguration, but the issue persists. I will solve that by using Slaesh's CC2652RB stick instead of SkyConnect.