Koenkk/zigbee2mqtt #16663: SONOFF Zigbee Dongle-P doesn't stay connected as a ROUTER - goes offline

jcat59, 2023-02-13:
### What happened?

I flashed my dongle successfully with the below command and had it working as a router. The problem is that after awhile (from a few minutes to a few hours) the router will go offline and no longer be responsive. From my research - the way to fix this is to configure the device to report periodically to the coordinator (around 4 minutes). I’m not sure if this is related or true or not. The router has never stayed online enough for any device to connect to it (other than coordinator).

Is there some way I can update the device to report to the coordinator so it doesn't go offline? If I have to reflash, can you guide me how I can alter the command line below? -- -command line uses a script here https://github.com/JelmerT/cc2538-bsl/issues/134 to load the firmware to my sonoff.

python cc2538-bsl.py -p COM3 -evw --ieee-address 00:12:4b:00:01:4d:12:34 --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_router_20221102.hex

I'm using ZHA in home assistant just in case that makes any difference.

### Zigbee2MQTT version

dfgdfg

### Adapter firmware version

dfgfdg

### Adapter

dfgdfg

closed 2023-03-23 by github-actions[bot]

teskanoo, 2023-09-23:
I am experiencing the same - the ZHA network map in homeassistant indicates many connections to the dongle but the dongle itself shows as offline

[image]

Ramonix07, 2024-05-01:
I have the same problem and I don't know how to solve it. Update sign to CC1352P2_CC2652P_launchpad_router_20221102.hex and sonoff zigbee 3.0 usb dongle plus.

Ramonix07, 2024-05-01:
This is a router sonoff zigbee 3.0 usb dongle plus, with firmware CC1352P2_CC2652P_launchpad_router_20221102.hex
[image]