Koenkk/zigbee2mqtt #16666: [New device support]: vendor: 'LELLKI', model: 'WK35' 'Power socket EU BR IT JP FR IL (with power monitoring)'

gargamelonly, 2023-02-13:
### Link

https://www.aliexpress.com/item/4001243518512.html?spm=a2g0o.order_list.order_list_main.11.7bb41802exmRvK

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0xa4c138641103ae0a","nwkAddr":53596,"manufId":4417,"manufName":"_TZ3000_ko6v90pg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ko6v90pg","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":232,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2669683,"53251":125,"53252":11332,"53253":1348,"53264":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1676328670898,"defaultSendRequestWhen":"immediate"}

### Comments

I tried adding an external converter similar to some existing one but is crashes z2m.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
	fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_ko6v90pg'}],
    model: 'TS011F_plug',
	description: 'Power socket EU BR IT JP FR IL (with power monitoring)',
    vendor: 'TuYa',
	whiteLabel: [{vendor: 'LELLKI', model: 'WK35'}],
    fromZigbee: [fz.on_off,	fz.electrical_measurement, fz.metering,	fz.ignore_basic_report,	fz.tuya_switch_power_outage_memory],
	toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff, 'haElectricalMeasurement', 'seMetering']);
	    await reporting.rmsVoltage(endpoint, {change: 5});
	    await reporting.rmsCurrent(endpoint, {change: 50});
	    await reporting.activePower(endpoint, {change: 10});
	    await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
};

module.exports = definition;
```

Koenkk, 2023-02-14:
This device should already work out-of-the-box with z2m 1.30.1

gargamelonly, 2023-02-14:
True. Sorry for not updating z2m.

closed 2023-02-15 by Koenkk