Koenkk/zigbee2mqtt #16668: 10+ devices just dropped off Network

cloudbr34k84, 2023-02-14:
### What happened?

Devices have just started dropping off the network. I have not changed anything. 

### Zigbee2MQTT version

Zigbee2MQTT Edge

### Adapter firmware version

20220219

### Adapter

Sonoff Coordinator P

### Debug log

[log bunch of devices just go offline.txt](https://github.com/Koenkk/zigbee2mqtt/files/10727389/log.bunch.of.devices.just.go.offline.txt)

Koenkk, 2023-02-14:
`SRSP - AF - dataRequest after 6000ms` indicates that your stick disconnects. There are many causes of this but usually other processes interfering, bad power supply or other usb devices interfering. Changing from a USB 3 to USB 2 port can also help.

cloudbr34k84, 2023-02-14:
@Koenkk Thanks for the tips. 
Its on USB2 already
Power is supplied via NUC and 5 meter USB extension cable. maybe thats tooo long?
its a few meters away anything else?? 

What do you mean by "other Processes"

It doesn't happen all the time.

Koenkk, 2023-02-15:
5 meter may be too long, try a shorter one (50cm)

cloudbr34k84, 2023-02-16:
> 5 meter may be too long, try a shorter one (50cm)

Ok yeah ill try shorter

closed 2023-02-21 by cloudbr34k84