Koenkk/zigbee2mqtt #16675: Stream or a document separator is expected (2:14)

Deejaybaze, 2023-02-14:
### What happened?

thomas@nuc-home:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.30.1 start
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: end of the stream or a document separator is expected (2:14)

 1 | .# Home Assistant integration (MQTT discovery).
 2 | homeassistant: false
------------------^
 3 |
 4 | # allow new devices to join
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1645:5)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)
    at read (/opt/zigbee2mqtt/lib/util/settings.ts:328:20)
    at getInternalSettings (/opt/zigbee2mqtt/lib/util/settings.ts:433:21)
    at Object.reRead (/opt/zigbee2mqtt/lib/util/settings.ts:716:5)
    at start (/opt/zigbee2mqtt/index.js:91:14)

### What did you expect to happen?

Start Z2M

### How to reproduce it (minimal and precise)

In folder type "npm start"

### Zigbee2MQTT version

new install

### Adapter firmware version

latest

### Adapter

SMLIGHT SLZB-06

Deejaybaze, 2023-02-14:
.# Home Assistant integration (MQTT discovery).
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://***************'
  # MQTT server authentication, uncomment if required:
  user: **********
  password: ***********

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: 'tcp://192.168.0.42:6638'

advanced:
  Network_key: GENERATE

frontend: true
port: *****
auth_token: *********

Koenkk, 2023-02-14:
Your yaml file is invalid, use https://www.yamllint.com to validate it.

closed 2023-02-14 by Koenkk

Deejaybaze, 2023-02-14:
That file is from the installation. I have not edited that. Can i replace it with a file downloadable from you?