Koenkk/zigbee2mqtt #16681: TS0601 Dimmer not controllable via Zigbee2MQTT Group

jzielke84, 2023-02-14:
### What happened?

Successfully added the light dimmer. Controllable via Z2M but not when added to a Z2M Group with other lights and switches.

### What did you expect to happen?

Being able to control it within the group.

### How to reproduce it (minimal and precise)

Create group and add all other lights and switches to the group > works fine (switch in HA is a normal on off slider)
Create group and add only this dimmer > group works fine (switch in HA is a lightning bolt symbol)
Add Dimmer to group with all other switches added already > Can toggle all other switches but dimmer doesn't react

When I open the group in Z2M and click the switch for the dimmer manually it works.

### Zigbee2MQTT version

1.30.1-dev commit: 6e193ca

### Adapter firmware version

20221102

### Adapter

Sonoff USB 3.0 P-Version

jzielke84, 2023-02-15:
Small notice on this issue: There is no refresh-symbol in Z2M regarding the dimmer's power state. Thats probably why HA shows two lightning bolts instead of a slider. Probably that's why the group fails.

[image]

Koenkk, 2023-02-15:
The TuYa device talk a different protocol, therefore they cannot work in groups where standard Zigbee commands are send to.

jzielke84, 2023-02-15:
Oh ok, thanks for clarifying. So I’ll stick with a helper group in HA combining the Z2M group and this single device. Still better than sending individual commands to every light when turning on and off large groups of lights.

closed 2023-02-15 by jzielke84