Koenkk/zigbee2mqtt #16684: zigbee2mqtt will not start after update.sh ran

SteEllis, 2023-02-15:
### What happened?

updated zigbee2mqtt using the update.sh script
after completion zigbee2mqtt was not running
checked for adapter and everything appeared correct
tried a reboot
tried to manually run with rpm start to get error log below
ensured npm collected all modules by issuing npm up
no change same error

### What did you expect to happen?

Expected zigbee2mqtt service to start

### How to reproduce it (minimal and precise)

issue npm start and error is visible

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

cannot find requested information frontend not started

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_0e9e20d7c912ec11b43221c7bd930c07-if00-port0 

### Debug log

> zigbee2mqtt@1.30.1 start
> node index.js

/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/ignore-enoent.js:15
        if (er?.code !== 'ENOENT') {
               ^

SyntaxError: Unexpected token '.'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/rimraf-posix.js:14:28)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/zigbee-user/.npm/_logs/2023-02-15T01_45_41_230Z-debug.log

Bolten88, 2023-02-15:
I've got the same error after updating to 1.30.1. Have you found a solution yet?

Bolten88, 2023-02-15:
I have reverted to version 1.28.0 and then it was working again. 

Is it related to my raspbian os version?

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Bolten88, 2023-02-15:
I did some trial and error with the versions and i was able to get it work on version 1.29.2. When i go to 1.30.* it fails to start with the above error.

SteEllis, 2023-02-15:
I did a fresh install and restored the zigbee2mqtt data. This now works with 1.30.1 as such I suspect something with the upgrade, npm packages?

Bolten88, 2023-02-15:
> I did a fresh install and restored the zigbee2mqtt data. This now works with 1.30.1 as such I suspect something with the upgrade, npm packages?

A fresh install of zigbee2mqtt only or did you reinstall the os as well? I’m aware now that i’m using a legacy os. So i thought maybe it is related.

Koenkk, 2023-02-15:
You are probably using an outdated node version, try upgrading to node 18 (run `node --version` to get the current version, min required is 14)

SteEllis, 2023-02-15:
> 

I did a full install including a the OS (raspberry pi)

Bolten88, 2023-02-15:
> You are probably using an outdated node version, try upgrading to node 18 (run `node --version` to get the current version, min required is 14)

$ node --version
v12.22.12

so I need to update that first. Will try that tomorrow.

nilsx, 2023-02-16:
upgraded node to v19.6.0 today, but still having the same problem

wantuchm, 2023-02-16:
I have updated nodejs from v12 to v14.21.2 and it is working for me. On node v12 I have same problem.

Bolten88, 2023-02-16:
> I have updated nodejs from v12 to v14.21.2 and it is working for me. On node v12 I have same problem.

How did you update to that specific version?

chrismast, 2023-02-16:
> nodejs

You can follow Option 2 outlined [here](https://phoenixnap.com/kb/update-node-js-version)

chrismast, 2023-02-17:
@Bolten88 : Have a look in my open (now closed) ticket. I got it working, maybe the steps help you as well.

elektron44, 2023-03-18:
> I have updated nodejs from v12 to v14.21.2 and it is working for me. On node v12 I have same problem.

This worked for me too.
Thanks guys

closed 2023-04-26 by github-actions[bot]