Koenkk/zigbee2mqtt #16685: Odd issue with IKEA Tradfri LED2003G10 bulbs

aatifk, 2023-02-15:
### What happened?

I have four of these bulbs controlled via a Home Assistant automation, which is supposed to turn the bulbs off if an associated motion sensor doesn't detect motion for a couple of minutes. At least once every 24 hours, the automation fires and the bulbs report in Zigbee2MQTT as off, except they're not actually off. If I force the state of bulbs to refresh using the refresh button in Z2M front-end, the state actually changes to showing on. There is no pattern as to what causes this except that it's been happening at least once every 24 hours, and with all four bulbs. 

[image]

### What did you expect to happen?

State of bulbs in Zigbee2MQTT should be reflecting the actual state. It shows "off" even when the bulbs are actually on. Forcing state to refresh in front-end updates the bulbs to start showing on.

### How to reproduce it (minimal and precise)

Happening randomly, no steps to reproduce. 

### Zigbee2MQTT version

1.30.1 commit: eb878d3d

### Adapter firmware version

20221226

### Adapter

Sonoff USB Zigbee Dongle Type P

Dielee, 2023-02-17:
Same issue here! For me the lamps are shown "on" but they are off. This happens once a day.
Any help ? Should we send debug logs ?

closed 2023-03-27 by github-actions[bot]

Stooovie, 2023-11-19:
Any more info? I have similar issues