Koenkk/zigbee2mqtt #16688: Tuya Smart Switch (No Neutral) Leaves network immediately after successful join.

lavinir, 2023-02-15:
### What happened?

Have tried two different smart wall switches (2-gang, No Neutral). Have a lot of these with Neutral around the house working perfectly. 
Every time I attempt to pair these, they go through the proper motions and are successfully configured only to leave the network a few seconds afterwards. This happens each time no matter how many attempts at repairing I make.
Logs:

```
49,33: Info 2023-02-15 09:41:33Device '0xa4c1385a27d3b9e0' joined
  50,108: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  50,144: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  51,48: Info 2023-02-15 09:41:33Starting interview of '0xa4c1385a27d3b9e0'
  52,108: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  52,144: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  53,56: Debug 2023-02-15 09:41:38Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
  54,56: Debug 2023-02-15 09:41:38Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
  55,56: Debug 2023-02-15 09:41:38Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  56,56: Debug 2023-02-15 09:41:40Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  57,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
  58,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
  59,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  64,51: Info 2023-02-15 09:41:42Successfully interviewed '0xa4c1385a27d3b9e0', device has successfully been paired
  65,33: Info 2023-02-15 09:41:42Device '0xa4c1385a27d3b9e0' is supported, identified as: TuYa Smart light switch - 2 gang (TS0012)
  66,887: ⟪ 686 characters skipped ⟫ngth)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0012","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"successful","supported":true},"type":"device_interview"}'
  66,923: ⟪ 686 characters skipped ⟫ngth)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0012","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"successful","supported":true},"type":"device_interview"}'
  67,38: Info 2023-02-15 09:41:42Configuring '0xa4c1385a27d3b9e0'
  68,56: Debug 2023-02-15 09:41:43Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":80,"manufacturerName":"_TZ3000_ewtuosug","modelId":"TS0012","powerSource":3,"zclVersion":3}' from endpoint 1 with groupID 0
  69,50: Info 2023-02-15 09:41:43Successfully configured '0xa4c1385a27d3b9e0'
  70,56: Debug 2023-02-15 09:41:48Received Zigbee message from '0xa4c1385a27d3b9e0', type 'attributeReport', cluster 'genBasic', data '{"65506":55,"65508":1,"appVersion":80}' from endpoint 1 with groupID 0
  71,36: Warning 2023-02-15 09:41:55Device '0xa4c1385a27d3b9e0' left the network
  72,108: Info 2023-02-15 09:41:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'
  72,144: Info 2023-02-15 09:41:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'
  75,33: Info 2023-02-15 09:41:59Device '0xa4c1385a27d3b9e0' joined
  76,108: Info 2023-02-15 09:41:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  76,144: Info 2023-02-15 09:41:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  77,48: Info 2023-02-15 09:41:59Starting interview of '0xa4c1385a27d3b9e0'
  78,108: Info 2023-02-15 09:41:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  78,144: Info 2023-02-15 09:41:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  92,36: Warning 2023-02-15 09:42:01Device '0xa4c1385a27d3b9e0' left the network
  93,108: Info 2023-02-15 09:42:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'
  93,144: Info 2023-02-15 09:42:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'
```

### What did you expect to happen?

Device stays on network.

### How to reproduce it (minimal and precise)

Pair the device.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZB Dongle-E

### Debug log

49,33: Info 2023-02-15 09:41:33Device '0xa4c1385a27d3b9e0' joined
  50,108: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  50,144: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_joined"}'
  51,48: Info 2023-02-15 09:41:33Starting interview of '0xa4c1385a27d3b9e0'
  52,108: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  52,144: Info 2023-02-15 09:41:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
[... 2 lines cut ...]
  55,56: Debug 2023-02-15 09:41:38Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  56,56: Debug 2023-02-15 09:41:40Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  57,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
  58,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
  59,56: Debug 2023-02-15 09:41:41Received Zigbee message from '0xa4c1385a27d3b9e0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
  64,51: Info 2023-02-15 09:41:42Successfully interviewed '0xa4c1385a27d3b9e0', device has successfully been paired
[... cut ...]
  78,144: Info 2023-02-15 09:41:59MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0","status":"started"},"type":"device_interview"}'
  92,36: Warning 2023-02-15 09:42:01Device '0xa4c1385a27d3b9e0' left the network
  93,108: Info 2023-02-15 09:42:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'
  93,144: Info 2023-02-15 09:42:01MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385a27d3b9e0","ieee_address":"0xa4c1385a27d3b9e0"},"type":"device_leave"}'

kirovilya, 2023-02-15:
try with latest/edge dev version zigbee2mqtt

lavinir, 2023-02-15:
Thanks for the reply @kirovilya . I can try but the reason I was holding back was an issue with my adapter that's also open here where mass devices drop and there are issues with adapter initialization (Sonoff ZB Dongle-E) that started to happen from 1.29. 

I guess I'll give it a shot and worst case will restore from backup.

jzielke84, 2023-02-15:
I noticed a similar behaviour with these [1 gang switch module](https://www.zigbee2mqtt.io/devices/TS0001_switch_module.html#tuya-ts0001_switch_module). When I tried to pair them after changing my network settings (channel, key etc.) by turning them on and off four times within one or two seconds, they were interviewed and suddenly left. Successful pairing was possible by holding the pairing-button for a couple of seconds (make sure they don't blink from an unfinished pairing process, they will stop after one or two minutes). Probably the four times on and off procedure is just a soft-reset (repair with same key/channel).

lavinir, 2023-02-15:
I will try going with the Edge version. I'm sure I'm doing the pariing properly with the device.. The other modules I have with N wire use the same process and pair perfectly.

I will update on the rsults from the edge version.

lavinir, 2023-02-16:
Unfortunately edge version has not resolved the issue.

kirovilya, 2023-02-16:
@lavinir can you record herdsman log? https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

lavinir, 2023-02-17:
Hey @kirovilya 
Thanks sure. Hope this helps:
[zigbee-herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/10766357/zigbee-herdsman.log)

kirovilya, 2023-02-17:
@lavinir thank you. I will deal

lavinir, 2023-02-18:
Thank you @kirovilya ! 
Btw I assume you figured it out yourself but forgot to add the device id is : a4c1385a27d3b9e0

lavinir, 2023-03-21:
Hi @kirovilya , were you able to make any progress with this ?

lavinir, 2023-03-23:
@kirovilya , Yes I'm in 1.30.2-1 
Device completes interview and configuration stages successfully then a few seconds afterwards I get the device has left the network message.

lavinir, 2023-04-27:
@kirovilya - Would any more information be helpful here ?

kirovilya, 2023-04-27:
@lavinir if updating z2m does not help, then you can try updating the sonoff stick firmware, for example, to 7.1.1 or 7.1.4
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

closed 2023-06-06 by github-actions[bot]