Koenkk/zigbee2mqtt #16689: Update from 1.22.2 to latest Version - Error starting z2mqtt

chrismast, 2023-02-15:
### What happened?

Hi all,
I do run z2mqtt since a long time, missed though to update since early last year.
I am still on version 1.22.2.
Wanted to update today to the latest version and followed the update instructions (as I did also before always) but somehow z2mqtt does not start anymore and I am not really getting any error message telling me why (or I cannot find the correct log).
I also looked through all releases and did not see any breaking changes.

**Below is my config:**

Running z2mqtt inside an LXC on Proxmox
npm --version: 6.14.15
node --version: v12.22.7

configuration.yaml:

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt.USER.com:1883'
  user: USER
  password: 'PW'
  #key: /home/christian/ssl/mqtt.key
  #cert: /home/christian/ssl/mqtt.crt
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: debug
  last_seen: ISO_8601_local
  cache_state: true
  cache_state_persistent: true
  cache_state_send_on_startup: true
  ikea_ota_use_test_url: true
  rtscts: false
experimental:
  output: attribute
frontend:
  port: 8080
  host: 0.0.0.0
devices: devices.yaml

```

Any hint on what could go wrong or where I could dig further on the error I am getting?
Luckily I can easily rollback with Proxmox so that I do not break the PROD environment.

Thanks

### What did you expect to happen?

Update as usual

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

zzh!

lux73, 2023-02-15:
upgrade nodeJS to at least v16.x

chrismast, 2023-02-15:
> upgrade nodeJS to at least v16.x
> 
> your Coordinator is running very old Firmware - consider upgrade too

Thanks for your quick reply.
Sadly somehow this also did not work.
I updated nodeJS to 18.14.0, afterwards did a npm rebuild and then followed the update cmds.
I still get an error restarting z2mqtt (below some log output):

```

Feb 15 19:08:28 mosquitto npm[8247]:     at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/dist/cjs/src/rimraf-posix.js:14:28)
Feb 15 19:08:28 mosquitto npm[8247]:     at Module._compile (internal/modules/cjs/loader.js:999:30)
Feb 15 19:08:28 mosquitto npm[8247]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! code ELIFECYCLE
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! errno 1
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! zigbee2mqtt@1.30.1 start: `node index.js`
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! Exit status 1
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR!
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! Failed at the zigbee2mqtt@1.30.1 start script.
Feb 15 19:08:28 mosquitto npm[8247]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

lux73, 2023-02-15:
pls try update npm to v8.x

is this even not work pls save your /data Directory inside zigbee2mqtt Folder and try a complete new installation.

afterwards you can copy your /data Folder back

found this related to the `npm ERR! errno 1`  https://sebhastian.com/npm-err-code-1/

Bolten88, 2023-02-16:
We are discussing this in this issue as well: 16689
You can downgrade zigbee2mqtt to get working again. But i'm also lookin in to how to update npm/nodejs to the right versions.

chrismast, 2023-02-16:
> We are discussing this in this issue as well: 16689 You can downgrade zigbee2mqtt to get working again. But i'm also lookin in to how to update npm/nodejs to the right versions.

Thanks guys!
Did not yet try to update nodejs/npm as I wanted to do this on the weekend.
I can easily revert back to an earlier snapshot via Proxmox, hence its not really an issue, can get it back working in a few seconds.
Good to know though that I am not the only one, might be easiest then maybe to redo from scratch.

chrismast, 2023-02-17:
Ok all, thanks again for your help!
I got it working with @lux73 latest post.

To update from 1.22.2 to 1.30.1 I followed:

- Update nodejs via Option 2 [here](https://phoenixnap.com/kb/update-node-js-version)
- Update npm via @lux73 's post with `npm install -g npm@9.3.1`
- `systemctl stop zigbee2mqtt`
- Follow the [standard update instructions](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl)
- `systemctl start zigbee2mqtt`

Thanks again!

chrismast, 2023-02-17:
Closing this issue.

closed 2023-02-17 by chrismast

lux73, 2023-02-17:
Nice! 😎