Koenkk/zigbee2mqtt #16692: Aqara E1 TRV OTA update installs wrong/dated version

programmdesign, 2023-02-15:
### What happened?

I used the OTA feature to update my Aqara E1 TRV from firmware version **0.0.0_0825** to a newer version. After the update completed, the firmware version is 0.0.0_0025:

[image]

When I check the version in the HA settings dialog, it also does not show an update from 0.0.0_0825 to a later version. It shows update from version 25 to 2329.

[image]

### What did you expect to happen?

I expected the TRV to be update to version [0.0.0_0925](https://www.aqara.com/en/version/radiator-thermostat-e1) and to see this version number also in the HA Settings Update dialogue. 

### How to reproduce it (minimal and precise)

Update the firmware of your Aqara E1 TRV via the Zigbee2MQTT UI.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20220219

### Adapter

zStack3x0

cwildfoerster, 2023-02-15:
Same Problem here

programmdesign, 2023-02-16:
@cwildfoerster - to my surprise, today the OTA page shows the TRV with the correct version number 0.0.0_0925. Not sure why, but is that also happening on your end?

[image]

cwildfoerster, 2023-02-16:
@programmdesign  Yes, it seems like a after some time it gets detected correctly. There was a commit saying `0.0.0_0025` was the default value if the correct one couldn't be determined. still have another one on 0.0.0_0825 tho, not yet updated after this.

macintosh-HD, 2023-02-21:
I have three Aqara TRVs. I used the Z2M update feature and updated all three from 0.0.0_0825. Two display 0.0.0_0925 and one 0.0.0_0025. All three also still want to update even though in the case of the 925 ones that is the official latest firmware at this point.

DenDeXTeR51, 2023-03-27:
Update to the latest-dev, and wait 1 hour after TRV upgrade - the version will be fixed.

Koenkk, 2023-03-28:
This should be fixed in the 1 April release indeed.

Mikaka27, 2024-03-10:
Hi, could you specify which commit/PR fixed this problem? Thanks.