Koenkk/zigbee2mqtt #16693: Problem with MLI Tint remote control 404049D

ChrisLSaar, 2023-02-15:
### What happened?

I have a problem with the integration of a MLI Tint remote control.

The model seems to be identical to the remote control 404049C, but has the type designation 404049D.

The debug log shows the familiar message of an unsupported device. Here you can see that the Zigbee model name should be ['Remote Control'] instead of the old name ['tint-Remote-white'].

I have created an external converter ...

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
      zigbeeModel: ['Remote Control'],
        model: '404022/404049C',
        description: 'Tint dim remote control',
        vendor: 'MLI',
        fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp,
            fz.command_move_to_color, fz.tint_scene],
        exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
            'brightness_stop', 'color_temperature_move', 'color_move', 'scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            device.powerSource = 'Battery';
            device.save();
        },
};

module.exports = definition;
```

With this converter I can integrate the remote control.

[image]

However, no actions are triggered.
In the debug log there is no more message at all for this participant, nevertheless the remote control does not work.

I have additionally created a debug log on level zigbee-herdsman*. However, I personally cannot do anything with this log.
I have attached the log.

[ZigbeeHerdsmanLog_404049D.txt](https://github.com/Koenkk/zigbee2mqtt/files/10746659/ZigbeeHerdsmanLog_404049D.txt)

Best regards
Christian

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20180815

### Adapter

CC2531

### Debug log

[ZigbeeHerdsmanLog_404049D.txt](https://github.com/Koenkk/zigbee2mqtt/files/10746678/ZigbeeHerdsmanLog_404049D.txt)

Koenkk, 2023-02-16:
Can you try this change:

```js
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1); // <- added 
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingColorCtrl']); // <- added 
            device.powerSource = 'Battery';
            device.save();
        },
```

ChrisLSaar, 2023-02-16:
Hi, yes, tried it directly.
Thanks for the quick feedback.
I have adjusted the external converter as you described.
There are now more debug outputs.
At the end of the pairing there is now an error.
Actions still don't come up.

Attached the new log...

[Zigbee2Mqtt_Log_404049D_new_converter.txt](https://github.com/Koenkk/zigbee2mqtt/files/10759842/Zigbee2Mqtt_Log_404049D_new_converter.txt)

Enemyscout, 2023-02-17:
Did a quick test. This seems to work for me. 
Device shows up an generates actions when buttons are pressed

ChrisLSaar, 2023-02-17:
@Enemyscout : 
Are you sure you have the same type of remote as me? 
404049D and not 404049C ?

The remote still does not work for me. No actions are triggered.

Koenkk, 2023-02-18:
@ChrisLSaar it may help to re-pair the device close to the coordinator. First force remove it through the z2m frontend, restart z2m and re-pair.

ChrisLSaar, 2023-02-18:
@Koenkk 

I have already tried this several times.
- Forced deletion of the device
- reboot the Zigbee2Mqqt container (also already rebooted the whole server)
- batteries out of the remote, wait, and batteries in again
- reintegrate the remote very close to the coordinator (about 30cm)

Always the same result...
The integration process seems to work but no actions are triggered. 
After some time a LQI is displayed for the device. But actions do not come up.

[image]

In the debug log, nothing is entered when a key is pressed.
Can I send you some more detailed logs that might help you?
Is there a possibility to log more deep?
Does the log from the zigbee-herdsman* help you?

Koenkk, 2023-02-19:
- Does the configure succeed? You can force it via the z2m frontend -> device -> yellow refresh icon. If it fails, try to wakeup the device right before pressing the button
- Can you provide the data/database.db entry of your device?

Koenkk, 2023-02-20:
I cannot understand why my converter doesn't work (especially since it worked for @Enemyscout). Nothing shows up in the debug log when pressing the buttons?

ChrisLSaar, 2023-02-20:
Yeah, I don't know either.
I am a software developer myself, but from a different field. 
Normally I shouldn't have been too stupid with the integration.
Two IKEA remote controls run without any problems.

What seems strange to me...
After some unused time the remote control it is displayed as offline.
[image]

When I press a key again the following debug messages are displayed...
[image]
[image]

After that the remote control is shown as ONLINE again,
but still no action triggered and no further debug output on any key press.

Koenkk, 2023-02-21:
@ChrisLSaar could it be that the remote is broken? (have you seen it working with any other system?)

ChrisLSaar, 2023-02-21:
I had ordered the remote just before my first post. But of course it is not excluded that it has a defect.
Maybe I'll try a Wireshark trace or I send the remote back and try another model.

Anyway, thanks for your efforts and your impressive project "Zigbee2Mqtt".
I will definitely "buy you some coffee".

ChrisLSaar, 2023-02-25:
I have sent remote control back to Amazon to not exceed return period.

closed 2023-02-25 by ChrisLSaar

Sir-Pandalot, 2023-08-07:
@Koenkk I just received this remote and I cannot get it to work. I added the below converter, but it still gives me this error:

Device '0xa4c13800aa432d18' with Zigbee model 'Remote Control' and manufacturer name 'MLI' is NOT supported

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'Remote Control', manufacturerName: 'MLI'}],
    model: '404049D',
    description: 'Tint dim and color remote control',
    vendor: 'Müller Licht',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp,
        fz.command_move_to_color, fz.tint_scene],
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'color_temperature_move', 'color_move', 'scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1); // <- added 
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingColorCtrl']); // <- added 
        device.powerSource = 'Battery';
        device.save();
    },
},

module.exports = definition;`