Koenkk/zigbee2mqtt #16700: Support for Tuya USB Zigbee Repeater TS0207

DaniCollado22, 2023-02-15:
### What happened?

The device is detected as https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector but should be https://www.zigbee2mqtt.io/devices/TS0207_repeater.html. This problem was noted here and fixed later: https://github.com/Koenkk/zigbee2mqtt/issues/4329

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

20190608

### Adapter

CC2531

### Debug log

{"id":25,"type":"Router","ieeeAddr":"0x1c34f1fffeb7701c","nwkAddr":3111,"manufId":4098,"manufName":"_TZ3000_misw04hq","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�v{+\u0013\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_misw04hq","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1676499632431}

lux73, 2023-02-16:
pls update your z2m to 1.30.1 and check if its not supported with the actual Version

1.18.1 is more than 1 Year old ^^

closed 2023-02-16 by Koenkk

mirkokg, 2023-10-20:
I am using: Z2M 1.33.1-dev commit: [b085fac](https://github.com/Koenkk/zigbee2mqtt/commit/b085fac)

and I am still experiencing this issue, I have three of these device, and all are being detected as 

TuYa: TS0207_water_leak_detector

[image]

mirkokg, 2023-10-20:
[image]

Koenkk, 2023-10-20:
@mirkokg please provide the data/database.db entry of your device.

mirkokg, 2023-10-20:
{
					"definition": {
						"description": "Water leak detector",
						"model": "TS0207_water_leak_detector",
						"supports": "water_leak, battery_low, battery, tamper, linkquality",
						"vendor": "TuYa"
					},
					"failed": [
					],
					"friendlyName": "Relay Hallway",
					"ieeeAddr": "0x943469fffe545f1d",
					"lastSeen": 1697818309764,
					"manufacturerName": "_TZ3000_nlsszmzl",
					"modelID": "TS0207",
					"networkAddress": 14391,
					"type": "Router"
				},
				{
					"definition": {
						"description": "Water leak detector",
						"model": "TS0207_water_leak_detector",
						"supports": "water_leak, battery_low, battery, tamper, linkquality",
						"vendor": "TuYa"
					},
					"failed": [
					],
					"friendlyName": "0x943469fffe57225d",
					"ieeeAddr": "0x943469fffe57225d",
					"lastSeen": 1697818404440,
					"manufacturerName": "_TZ3000_nlsszmzl",
					"modelID": "TS0207",
					"networkAddress": 25890,
					"type": "Router"
				}

mirkokg, 2023-10-20:
```
{
	"appVersion": 64,
	"dateCode": "",
	"defaultSendRequestWhen": "immediate",
	"endpoints": {
		"1": {
			"binds": [
				{
					"cluster": 1,
					"deviceIeeeAddress": "0xe0798dfffeeadbc3",
					"endpointID": 1,
					"type": "endpoint"
				}
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"65503": "    ",
						"65506": 22,
						"appVersion": 64,
						"dateCode": "",
						"hwVersion": 1,
						"manufacturerName": "_TZ3000_nlsszmzl",
						"modelId": "TS0207",
						"powerSource": 1,
						"stackVersion": 0,
						"zclVersion": 3
					}
				}
			},
			"configuredReportings": [
			],
			"devId": 8,
			"epId": 1,
			"inClusterList": [
				0,
				10,
				3
			],
			"meta": {
			},
			"outClusterList": [
				25
			],
			"profId": 260
		}
	},
	"epList": [
		1
	],
	"hwVersion": 1,
	"id": 8,
	"ieeeAddr": "0x943469fffe57225d",
	"interviewCompleted": true,
	"lastSeen": 1697815908072,
	"manufId": 4098,
	"manufName": "_TZ3000_nlsszmzl",
	"meta": {
	},
	"modelId": "TS0207",
	"nwkAddr": 25890,
	"powerSource": "Mains (single phase)",
	"stackVersion": 0,
	"type": "Router",
	"zclVersion": 3
}
```

`{
	"appVersion": 64,
	"dateCode": "",
	"defaultSendRequestWhen": "immediate",
	"endpoints": {
		"1": {
			"binds": [
				{
					"cluster": 1,
					"deviceIeeeAddress": "0xe0798dfffeeadbc3",
					"endpointID": 1,
					"type": "endpoint"
				}
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"65503": "    ",
						"65506": 22,
						"appVersion": 64,
						"dateCode": "",
						"hwVersion": 1,
						"manufacturerName": "_TZ3000_nlsszmzl",
						"modelId": "TS0207",
						"powerSource": 1,
						"stackVersion": 0,
						"zclVersion": 3
					}
				}
			},
			"configuredReportings": [
			],
			"devId": 8,
			"epId": 1,
			"inClusterList": [
				0,
				10,
				3
			],
			"meta": {
			},
			"outClusterList": [
				25
			],
			"profId": 260
		}
	},
	"epList": [
		1
	],
	"hwVersion": 1,
	"id": 11,
	"ieeeAddr": "0x943469fffe545f1d",
	"interviewCompleted": true,
	"lastSeen": 1697815854333,
	"manufId": 4098,
	"manufName": "_TZ3000_nlsszmzl",
	"meta": {
	},
	"modelId": "TS0207",
	"nwkAddr": 14391,
	"powerSource": "Mains (single phase)",
	"stackVersion": 0,
	"type": "Router",
	"zclVersion": 3
}`

mirkokg, 2023-10-20:
This is how this device looks:

[image]

Koenkk, 2023-10-20:
`_TZ3000_nlsszmzl` will be detected correctly in the 1 November z2m version.

msmthng, 2024-10-11:
hi, iam running on Zigbee2MQTT-Version1.40.2 but the repeater is still recognized as Tuya | TS0207_water_leak_detector

and it throws this error al the time:

[2024-10-11 08:26:24] error: 	z2m: Failed to configure 'Repeater 2b', attempt 3 (Error: ZCL command 0x0cae5ffffe872ce6/1 genPowerCfg.configReport([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 10421 - 1 - 158 - 1 - 7 after 10000ms)

    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)

    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:640:47)

    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:672:28)

    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20

    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)

    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)

    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)

    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)

    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)

    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9))

is ther something i have to change or should it work? 

thx

FRahmann, 2024-10-26:
> z2m

same problem here. Also detected as wrong type

SoftwareSchlosser, 2025-02-08:
I just paired an actual [TS0207 water leak detector](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html) but it was detected as [eWeLink SNZB-03 motion sensor](https://www.zigbee2mqtt.io/devices/SNZB-03.html) ;)