Koenkk/zigbee2mqtt #16709: TuYa BW-SHP13 "TS011F_plug_1" - False zero Watt power data

SaturnusDJ, 2023-02-16:
### What happened?

(Opening this as problem because incorrect information is at play.)

Got two TuYa BW-SHP13 "TS011F_plug_1" today. I actually was not surprised to see that this looks the same as the ten I got almost three years ago, yet have different internals... Basic TuYa politics...

And this different version thus comes with a problem:
**The power and current attributes drop to 0 (W, A) every 4.5 minutes. Those are false readings.**

```
22:01:20: '{"child_lock":"UNLOCK","current":0.03,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:01:25: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:04:20: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:04:20: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:05:18: '{"child_lock":"UNLOCK","current":0,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:05:18: '{"child_lock":"UNLOCK","current":0,"energy":0.3,"indicator_mode":"off/on","linkquality":138,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:05:23: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:05:23: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:41: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:42: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:46: '{"child_lock":"UNLOCK","current":0,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":13,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:46: '{"child_lock":"UNLOCK","current":0,"energy":0.3,"indicator_mode":"off/on","linkquality":138,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:51: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":0,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
22:09:54: '{"child_lock":"UNLOCK","current":0.09,"energy":0.3,"indicator_mode":"off/on","linkquality":134,"power":12,"power_outage_memory":"restore","state":"ON","update":{"state":"idle"},"voltage":234}'
```
(Cut out some column for readability)

Can we filter this? Maybe a device_specific option to only accept zero if it is zero for x seconds. Where x should be 6 for the log above.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

zzh! CC2652R

ceesvanegmond, 2023-03-04:
@SaturnusDJ Having the same problem It seems to be hardware related issue https://github.com/Koenkk/zigbee2mqtt/issues/12421

ceesvanegmond, 2023-03-04:
@SaturnusDJ Having these kind of graphs at the moment 😆 
[image]

SaturnusDJ, 2023-03-04:
That is truly nuts...
I get the feeling these are not hardware defects but bad design. Unless there are users with no dips? Please post.

Therefore I hope it is possible to add a setting to only accept zero values after three zero values have been reported.

SaturnusDJ, 2023-03-04:
I reported the issue to BlitzWolf now.

ceesvanegmond, 2023-03-05:
@SaturnusDJ Yeah; they probably aren't going to do anything with it... Unfortunately...

SaturnusDJ, 2023-03-07:
First they asked where I bought them, then they told me to contact that platform for a restock date. Completely irrelevant replies. What an idiots. Guess BlitzWolf does nothing more than put a label on boxes. Will try TuYa.

SaturnusDJ, 2023-03-18:
TuYa also can't help because by buying from them, BlitzWolf takes all responsibility/accountability I guess.
I wrote again to BlitzWolf. After that I will mention Koen here, hopefully he is open to a workaround. Otherwise it's game-over.

SaturnusDJ, 2023-03-27:
@Koenkk 
Can you share your idea of what this is and what to do with it?
BlitzWolf is unwilling to look at their firmware.

Koenkk, 2023-03-28:
My expectation is that the firmware is bugged and incorrectly reports 0, to confirm, please provide the debug log when zero is reported.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

vincentscode, 2023-04-13:
The device does seem to report 0.
```
Debug 2023-04-13 22:12:04	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":69}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:06	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":65}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:11	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:13	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":66}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:22	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":69}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:27	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":65}' from endpoint 1 with groupID null
Debug 2023-04-13 22:12:30	Received Zigbee message from 'Server Socket', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":63}' from endpoint 1 with groupID null
```

Koenkk, 2023-04-14:
I see, I cannot fix it from the z2m side, it's an issue of the device itself.

vincentscode, 2023-04-14:
That's unfortunate, but understandable. I assume contacting some tuya whitelabel company will have little effect.
Is there any way to filter this in z2m?

Koenkk, 2023-04-14:
@vincentscode how do you want to filter it? You don't know wether the `0` report is a false report or a real measurement.

vincentscode, 2023-04-14:
@Koenkk Sure, but it seems like it's a very short singular measurement (<=2s until next message) and is surrounded by wildly different values. This seems like something that would never occur in real-world use.

Koenkk, 2023-04-14:
The `debounce` option might do the trick: https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options

vincentscode, 2023-04-14:
Unfortunately using a debounce value just causes it to send no data at all to Home Assistant.
[image]

Koenkk, 2023-04-15:
@vincentscode can you provide the debug log when load is removed from the plug (so 0 is the real value)? What we can can is e.g. always ignore the first 0 value

vincentscode, 2023-04-15:
@Koenkk Sure. I've not removed the other attributeReports this time:
```
Debug 2023-04-15 09:28:09Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:09Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:10Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:19Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:19Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:20Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:21Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":227}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:28Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":226}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:29Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:29Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:30Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:33Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":227}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:39Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:39Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:40Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:42Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":226}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:47Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":225}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:49Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:49Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID null
Debug 2023-04-15 09:28:50Received Zigbee message from 'Test Socket 2', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
```

It seems from the log that removing the first 0 (or having the option to) should work fine as long as the reporting frequency is set high enough.

Koenkk, 2023-04-15:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/9a00f014c9b4c4588a49db6f49cfe0be

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed (you will see `TEMPORARY LOG: SKIPPING 0 VALUE` in the log)

vincentscode, 2023-04-15:
Thanks, this seems to be working! I do get the log messages about as frequently as I was getting the 0 data so it appears to be filtering correctly.

While we're at it: The same issue seems to also affect current measurement (which makes sense i suppose):
[image]

Voltage also drops to zero sometimes for 1 dataframe but it's a lot less frequent:
[image]

Additionally I've noticed that when testing with a very low intermittent load (such as a laptop loading with a full battery) "TEMPORARY LOG: SKIPPING 0 VALUE" get's spammed quite a lot and no data is reported, I am unsure as to why that's the case but it seems like it likely won't be an issue for most people.

Koenkk, 2023-04-15:
Updated https://gist.github.com/Koenkk/9a00f014c9b4c4588a49db6f49cfe0be to also apply the fix to current and voltage, can you check it?

Note that the `TEMPORARY LOG: SKIPPING 0 VALUE` logging will be removed in the final fix.

vincentscode, 2023-04-15:
It seems to be working just fine with the new version.

Koenkk, 2023-04-16:
@vincentscode @SaturnusDJ can you both provide the data/database.db entry of your device? Then I will make sure this fix only gets applied to your device version.

vincentscode, 2023-04-16:
I've got these two:
```
{"id":8,"type":"Router","ieeeAddr":"0xa4c1389e67732ef2","nwkAddr":32527,"manufId":4417,"manufName":"_TZ3000_gvn91tmx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0005L�+\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gvn91tmx","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":225,"rmsCurrent":342,"activePower":67}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,1437]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2862919,"53251":121,"53252":10714,"53253":1257,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":2820,"attrId":1291,"minRepIntval":1,"maxRepIntval":10,"repChange":1},{"cluster":2820,"attrId":1288,"minRepIntval":1,"maxRepIntval":10,"repChange":1},{"cluster":2820,"attrId":1285,"minRepIntval":1,"maxRepIntval":10,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1681640527425,"defaultSendRequestWhen":"immediate"}
{"id":9,"type":"Router","ieeeAddr":"0xa4c1385a9b06c337","nwkAddr":46847,"manufId":4417,"manufName":"_TZ3000_gvn91tmx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_gvn91tmx","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":226,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2767245,"53251":122,"53252":11118,"53253":1300,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff051b33","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":10,"repChange":1},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":10,"repChange":1},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":10,"repChange":1},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1681546683144,"defaultSendRequestWhen":"immediate"}
̀ ``

SaturnusDJ, 2023-04-16:
Hey guys, sorry I was away. Thanks for the progress. 👍🏼👌🏼

Currently also two in the network here:
```
{"id":53,"type":"Router","ieeeAddr":"0x","nwkAddr":14527,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0013W�+\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":233,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,432]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2793982,"53251":133,"53252":9207,"53253":1288,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1681425489969,"defaultSendRequestWhen":"immediate"}
{"id":58,"type":"Router","ieeeAddr":"0x","nwkAddr":15544,"manufId":4417,"manufName":"_TZ3000_amdymr7l","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���+\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_amdymr7l","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":234,"rmsCurrent":492,"activePower":75}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,9300]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2581720,"53251":142,"53252":9987,"53253":1394,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1681645586163,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-04-16:
Integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

izb2000, 2023-04-18:
It does not work correctly. If you remove the load, power and current do not show 0, they continue to show the last load value.

stathismes, 2023-05-02:
I confirm this too. Last loaded value remains once you turn the power off.

PunyIron, 2023-05-02:
@Koenkk, I can confirm wrong behaviour too. Your modification made above works correctly only if non-zero power load is continuous. If zero load is occuring, reporting is wrong - it reports last non-zero value. So it is never possible to achieve zero reporting. Is it possible to repair it?
Thanks.

SaturnusDJ, 2023-05-02:
Maybe my initial suggestion would work?  
> Maybe a device_specific option to only accept zero if it is zero for x seconds. Where x should be 6 for the log above.
So basically to hold back on the zero for a few seconds because if it is a wrong zero, the plug soon enough will report >0. But if it does not, then we know that it really is zero.

starlightbg, 2023-05-02:
Version 1.30.4 broke _TZ3000_gvn91tmx on my side too. Now it shows the last value that is >0.
Even if the plug is off, it does not show "0".

{
    "child_lock": "UNLOCK",
    **"current": 8.46,**
    "energy": 56.32,
    "indicator_mode": "off",
    "last_seen": "2023-05-02T13:55:20.077Z",
    "linkquality": 164,
    **"power": 1582,**
    "power_outage_memory": "restore",
    **"state": "OFF",**
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "voltage": 228,
    "update_available": null
}

vincentscode, 2023-05-02:
I'm also seeing this bug @Koenkk. I think the error is a result of deleting the key (`delete result[key]`, L.733) before putting it into the `globalStore` (L.736).
https://github.com/Koenkk/zigbee-herdsman-converters/commit/4a4b156103a8790e199560bf72254315821ec6fd#diff-7ae23d3c13c2d7f7e37106590684c791aba8093f8be813b7c74a4f1484017c3bR733-R736

Koenkk, 2023-05-02:
@vincentscode applied the fix, thanks!

Can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

temasnick, 2023-05-02:
> @vincentscode applied the fix, thanks!
> 
> Can you check if it works?

It's working but delayed. To show 0 watt must be pass 5 to 10 mins. And works unreliably.

Thanks for your work.

squadramunter, 2023-05-03:
Can confirm. Still does have a hight delay in updating Wattage when from powererd off state tot turned on state.
[image]

Koenkk, 2023-05-03:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/c7bb3c0a01bf48fd1c9ee89aaf2316bb

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

temasnick, 2023-05-03:
@Koenkk, yes it's working with external converter. Only 5 to 8 secs delay for 0 watt to appear. Thanks...

starlightbg, 2023-05-03:
@Koenkk Still not ok, with the latest external converter..

[image]

squadramunter, 2023-05-03:
Same here running Home Assistant Z2M Edge version. Still not updating as quickly as usual. Still takes around 5 to 7 minutes even with external addon added in the configuration.

starlightbg, 2023-05-03:
More logs when powered on from zero:

"last_seen":"2023-05-03T19:14:48.769Z","linkquality":220,"power":0,"power_outage_memory":"restore","state":"OFF"
"last_seen":"2023-05-03T19:14:48.769Z","linkquality":220,"power":0,"power_outage_memory":"restore","state":"ON"
"last_seen":"2023-05-03T19:14:48.778Z","linkquality":220,"power":0,"power_outage_memory":"restore","state":"ON"
"last_seen":"2023-05-03T19:15:40.532Z","linkquality":220,"power":0,"power_outage_memory":"restore","state":"ON"
"last_seen":"2023-05-03T19:15:40.627Z","linkquality":220,"power":7,"power_outage_memory":"restore","state":"ON"
"last_seen":"2023-05-03T19:15:42.415Z","linkquality":220,"power":7,"power_outage_memory":"restore","state":"ON"

temasnick, 2023-05-03:
> If the "Min rep change" is more than the max value it not working.
> In my case min rep change is 10 , the val of the power is 7W

Isn't it expected behaviour? When active power below min rep change it is only tirigger reporting for time variable, am i right?
I installed my device couple days ago. I don't know expected behaviour.

starlightbg, 2023-05-03:
@temasnick you are correct, but in the first screenshot more that 1 minute passed since the plug was powered off and the power was still 7 , not 0 :)

mrdroopy, 2023-05-03:
javascript fix works like a charm here, though its all plugs doing >10W loads.

starlightbg, 2023-05-03:
I just noticed that on a plug with high load the power is set to 0 on power off after some seconds, bit the "current" keeps the last value.
[image]

squadramunter, 2023-05-04:
@mrdroopy Could you check with a lamp that uses less then 10W of power? Does it update as quickly or does it still have a delay? With my setup it has still a delay

squadramunter, 2023-05-04:
@starlightbg I would like to see that Power and Current is updating realtime or at least updating in seconds rather then minutes what it is doing now. I use a graph in Home Assistant that is time based and use automations when something turns on and off like my lights in house. Should be good to know it turned on the exact time off and on. Now I need to guess everytime when it turned on or off :-) 

I hope there is a solution to this problem.

Koenkk, 2023-05-04:
@starlightbg 

> Not sure if @Koenkk can implement some logic like :
if value is 0 -> wait for example 5 seconds -> if no new value arrives during this 5 second period -> set the original value
if new value arrives during this period to use it.

That is exactly what this ext converter should be doing. I've updated the converter with some logging so we can better see what is happening.  Please test https://gist.github.com/Koenkk/c7bb3c0a01bf48fd1c9ee89aaf2316bb

starlightbg, 2023-05-04:
Here is when i turn off the plug and the "current" keeps the last value for a long time. In this case 60 seconds

`info 2023-05-04 19:07:33 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:33.112Z","linkquality":200,"power":7,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
info 2023-05-04 19:07:45 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:45.237Z","linkquality":200,"power":7,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
info 2023-05-04 19:07:45 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:45.237Z","linkquality":200,"power":7,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
info 2023-05-04 19:07:45 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:45.246Z","linkquality":196,"power":7,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
warning 2023-05-04 19:07:48 TS011F_TEMP: Publishing power undefined immediately
warning 2023-05-04 19:07:48 TS011F_TEMP: Publishing voltage undefined immediately
info 2023-05-04 19:07:48 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:48.114Z","linkquality":204,"power":7,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
warning 2023-05-04 19:07:48 TS011F_TEMP: Clearing existing 0 current timer, new value is undefined
warning 2023-05-04 19:07:48 TS011F_TEMP: Publishing current undefined immediately
warning 2023-05-04 19:07:48 TS011F_TEMP: Publishing voltage undefined immediately
info 2023-05-04 19:07:48 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:48.126Z","linkquality":204,"power":7,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
warning 2023-05-04 19:07:53 TS011F_TEMP: Publishing power delayed 0 value with timer
info 2023-05-04 19:07:53 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:07:48.126Z","linkquality":204,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":228}'
info 2023-05-04 19:08:39 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:08:39.939Z","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
warning 2023-05-04 19:08:44 TS011F_TEMP: Publishing power delayed 0 value with timer
info 2023-05-04 19:08:44 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0.14,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:08:39.939Z","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
warning 2023-05-04 19:08:44 TS011F_TEMP: Publishing current delayed 0 value with timer
info 2023-05-04 19:08:44 MQTT publish: topic 'zigbee2mqtt/Kitchen Led', payload '{"child_lock":"UNLOCK","current":0,"energy":1.8,"indicator_mode":"off","last_seen":"2023-05-04T16:08:39.939Z","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'`

Koenkk, 2023-05-04:
@starlightbg think I found the bug, please check with https://gist.github.com/Koenkk/c7bb3c0a01bf48fd1c9ee89aaf2316bb

mrdroopy, 2023-05-04:
Power and current both return to 0 within seconds. Looks ok at first sight!

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

vincentscode, 2023-05-04:
[image]
I've had it drop to 0 momentarily now again (tho much less frequent) with the script from yesterday. I suppose it did not correct itself within the 5s window.

Koenkk, 2023-05-04:
@vincentscode you can play with the timeout, e.g. to use 10 sec change `, 5 * 1000));` to `, 10 * 1000));` in the ext converter

vincentscode, 2023-05-04:
I'm sure that will fix it. Just as a note for the final implementation.

starlightbg, 2023-05-04:
> @starlightbg think I found the bug, please check with https://gist.github.com/Koenkk/c7bb3c0a01bf48fd1c9ee89aaf2316bb

@Koenkk Working fine now. No "undefined" values.
Timeout of 5 seconds is ok in my opinion. If it is really a false zero and the device is on with load, it should report the correct value in those 5 seconds. If it does not something is totally wrong with it :)

mrfifiak, 2023-05-05:
I found this thread when searching for the reason I'm not getting notifications that my laundry is done - and can confirm that the fix in external converter works. 
[image]

squadramunter, 2023-05-05:
I don't know why but for me it's not working at all. When powering on Current is updating first and then a minute later Wattage is updating. When powering off Current is the first one that goes tot 0 and Wattage stays active longer as usual.

squadramunter, 2023-05-05:
So I tested it with my automation. The switch of the power plug turned ON at 15:20 but it reported power in Watt at 15:22

It turned OFF at 15:30 but reported to ZERO at 15:44

Conclusion the graph is still off compared to my automation traces.

I tested it with lights so I get always the exact value. It does not spike or anything.

[image]

Koenkk, 2023-05-05:
@squadramunter please post the debug log when using https://gist.github.com/Koenkk/c7bb3c0a01bf48fd1c9ee89aaf2316bb

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

squadramunter, 2023-05-05:
@Koenkk Thanks I will do. Report it to you soon

squadramunter, 2023-05-05:
`2023-05-05 21:06:27Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,5]}' from endpoint 1 with groupID null
2023-05-05 21:06:27Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":228}' from endpoint 1 with groupID null
2023-05-05 21:06:54Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
2023-05-05 21:07:19Received MQTT message on 'zigbee2mqtt/Woonkamer Lampen/set' with data 'ON'
2023-05-05 21:07:19Publishing 'set' 'state' to 'Woonkamer Lampen'
2023-05-05 21:07:19Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
2023-05-05 21:07:24Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":74}' from endpoint 1 with groupID null
2023-05-05 21:08:09Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":0}' from endpoint 1 with groupID null
2023-05-05 21:08:14Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":67}' from endpoint 1 with groupID null
2023-05-05 21:09:26Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null
2023-05-05 21:11:50Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,5]}' from endpoint 1 with groupID null
2023-05-05 21:11:50Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":74,"rmsVoltage":229}' from endpoint 1 with groupID null
2023-05-05 21:12:04Received Zigbee message from 'Woonkamer Lampen', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID null`

[98 more replies]

closed 2023-08-18 by github-actions[bot]