Koenkk/zigbee2mqtt #16715: SkyConnect & Aqara sensors

jvuotila, 2023-02-17:
### What happened?

Aqara sensors (end devices) like to join to zigbee mesh only thru coordinator which is SkyConnect. This is quite big problem because I lose connection on other end of my house. Philips motion sensors which I also got, use whole mesh. There is mainly Ikea on mesh.

Total 36
By device type
End devices: 21
Router: 15
By power source
Battery: 21
Mains (single phase): 15
By vendor
IKEA of Sweden: 18
LUMI: 13
Signify Netherlands B.V.: 4
Philips: 1
By model
TRADFRIbulbE27WSglobeopal1055lm: 6
lumi.sensor_magnet.aq2: 5
lumi.weather: 4
lumi.sensor_wleak.aq1: 4
TRADFRI on/off switch: 4
TRADFRI control outlet: 3
TRADFRIbulbE14WScandleopal470lm: 2
SML003: 2
TRADFRI Driver 30W: 2
SML004: 2
TRADFRI Driver 10W: 1
LWA001: 1

ID_MODEL: SkyConnect_v1.0
ID_MODEL_ENC: SkyConnect\x20v1.0
ID_MODEL_ID: ea60

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect

kirovilya, 2023-02-17:
try with latest dev/edge version of z2m

jvuotila, 2023-02-17:
Isn't that 1.30.1-1 latest release?

jvuotila, 2023-02-19:
Okay, just checked code of dev branch. There is some changes to curtain devices and almost daily update for zigbee-herdsman library. Just wondering that maybe I just wait for the next version then? Seems that there is no direct code changes to this issue.

mamrai1, 2023-02-22:
https://github.com/Koenkk/zigbee2mqtt/issues/16778

mamrai1, 2023-02-23:
Try to select the nearest router to your aqara sensor and select that for pairing... it actually works

jvuotila, 2023-02-23:
> Try to select the nearest router to your aqara sensor and select that for pairing... it actually works

Yep, that's incredible tip. I just paired half of my sensors again by this way and it worked perfectly!

closed 2023-04-03 by github-actions[bot]

DannyFeliz, 2023-05-07:
> Try to select the nearest router to your aqara sensor and select that for pairing... it actually works

This fixed my issue too, thanks.

Nailik, 2023-08-16:
How did you actually get it to work with skyconnect i struggle with the error mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/16716

mamrai1, 2023-08-16:
Ok, which firmware do you have?

Nailik, 2023-08-16:
[image]

I used this site https://skyconnect.home-assistant.io/firmware-update/

mamrai1, 2023-08-16:
Ok if you used zha or the multiprotocol plugin you have to disablecor uninstall  them first. After doing that can you probe the stick with 'universal-silabs-flasher --verbose --device /dev/ttyACM0 probe'

Nailik, 2023-08-16:
@mamrai1 thank you very much i only had to remove the multiprotocol plugin!