Koenkk/zigbee2mqtt #16716: Using skyconnect with Home Assistant

suniltaneja, 2023-02-17:
### What happened?

Zigbee2MQTT:error 2023-02-17 13:38:43: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:528:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:326:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

### What did you expect to happen?

Zigbee2MQTT does not start up

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

Nabu_Casa_SkyConnect_v1.0_12826d6552d5ec11aaf822b1a1ca919a

### Adapter

Skyconnect

En3myNightmare, 2023-02-18:
I had the same error. At first I thought it was the ZHA, which was already installed and I had actually removed completely. 
But it was due to the add-on "Silicon Labs Multiprotocol". This seems to have blocked access to the stick. 

Now my SkyConnect is connected to Z2M. But it runs anything but well. The devices are connected, but I get a timeout for every action. 

Z2M still does not seem to be SkyConnect compatible.

MikyM0use, 2023-02-26:
same problem here, I already put adapter: ezsp in the configuration file and checked the permission on the usb device... started mosquitto in a separate container, then used the following command to start zigbee2mqtt as docker (no home assistant involved at all) 

`sudo docker run    --name zigbee2mqtt    --restart=unless-stopped    --device=/dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_MY_ID-if00-port0:/dev/ttyACM0    -p 8080:8080    -v /home/michele/z2m/data:/app/data    -v /run/udev:/run/udev:ro    -e TZ=Europe/Rome    koenkk/zigbee2mqtt`

ends up in the same situation of the OP. 

>  Using '/app/data' as data directory
Zigbee2MQTT:info  2023-02-26 12:49:33: Logging to console and directory: '/app/data/log/2023-02-26.12-49-33' filename: log.txt
Zigbee2MQTT:info  2023-02-26 12:49:33: Starting Zigbee2MQTT version 1.30.1 (commit #eb878d3)
Zigbee2MQTT:info  2023-02-26 12:49:33: Starting zigbee-herdsman (0.14.89)
Zigbee2MQTT:error 2023-02-26 12:49:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-26 12:49:53: Failed to start zigbee
Zigbee2MQTT:error 2023-02-26 12:49:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-26 12:49:53: Exiting...
Zigbee2MQTT:error 2023-02-26 12:49:53: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:518:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:325:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)

closed 2023-04-06 by github-actions[bot]