Koenkk/zigbee2mqtt #16717: [New device support]:

philippsandhaus, 2023-02-17:
### Link

https://about.ikea.com/en/newsroom/2023/02/14/ikea-launches-vindstyrka-a-smart-sensor-to-measure-indoor-air-quality

### Database entry

{"id":40, "type":"Router", "ieeeAddr":"0x187a3efffe8104ed", "nwkAddr":18782, "manufId":4476, "manufName":"IKEA of Sweden", "powerSource":"Mains (single phase)", "modelId":"VINDSTYRKA", "epList":[1,242], "endpoints":{ 	"1":{ 		"profId":260, 		"epId":1, 		"devId":770, 		"inClusterList":[0,3,4,1026,1029,64599,64636,1066,64638], 		"outClusterList":[3,25,32,514], 		"clusters":{}, 		"binds":[], 		"configuredReportings":[], 		"meta":{} 	}, 	"242":{ 		"profId":41440, 		"epId":242, 		"devId":97, 		"inClusterList":[], 		"outClusterList":[33], 		"clusters":{}, 		"binds":[], 		"configuredReportings":[], 		"meta":	{}} 		}, 	"appVersion":16, 	"stackVersion":106, 	"hwVersion":1, 	"dateCode":"20220719", 	"swBuildId":"1.0.010", 	"zclVersion":8, 	"interviewCompleted":true, 	"meta":{}, 	"lastSeen":1676659177867, 	"defaultSendRequestWhen":"immediate" }

### Comments

I tried to add the device as described in the guide for adding new devices, but unfortunately the device does not report anything on its own. The guide suggests to add a configure: section and looking for definitions of similar devices, but I do not find any similar devices in Ikea.js. Could anyone point me in the right direction?

This is the entry in database.db
```json
{"id":40,
"type":"Router",
"ieeeAddr":"0x187a3efffe8104ed",
"nwkAddr":18782,
"manufId":4476,
"manufName":"IKEA of Sweden",
"powerSource":"Mains (single phase)",
"modelId":"VINDSTYRKA",
"epList":[1,242],
"endpoints":{
	"1":{
		"profId":260,
		"epId":1,
		"devId":770,
		"inClusterList":[0,3,4,1026,1029,64599,64636,1066,64638],
		"outClusterList":[3,25,32,514],
		"clusters":{},
		"binds":[],
		"configuredReportings":[],
		"meta":{}
	},
	"242":{
		"profId":41440,
		"epId":242,
		"devId":97,
		"inClusterList":[],
		"outClusterList":[33],
		"clusters":{},
		"binds":[],
		"configuredReportings":[],
		"meta":	{}}
		},
	"appVersion":16,
	"stackVersion":106,
	"hwVersion":1,
	"dateCode":"20220719",
	"swBuildId":"1.0.010",
	"zclVersion":8,
	"interviewCompleted":true,
	"meta":{},
	"lastSeen":1676659177867,
	"defaultSendRequestWhen":"immediate"
}
```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VINDSTYRKA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'VINDSTYRKA', // Vendor model number, look on the device for a model number
    vendor: 'IKEA of Sweden', // Vendor of the device (only used for documentation and startup logging)
    description: 'Humidity and air quality sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

peterkappelt, 2023-02-18:
I've got one, too. Temperature and Humidity reporting is fairly standard and works with the code below, for instance. PM2.5 and TVOC though seems to be IKEA specific. Does anynone happen to have a Dirigera Hub and is able to provide a Zigbee packet capture so we can extract some information?

```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ["VINDSTYRKA"],
  model: "VINDSTYRKA",
  vendor: "IKEA of Sweden",
  description: "IKEA Air Quality and Humidity sensor",
  fromZigbee: [fz.temperature, fz.humidity],
  toZigbee: [],
  exposes: [e.temperature(), e.humidity()],
  configure: async (device, coordinatorEndpoint, logger) => {
    const ep = device.getEndpoint(1);
    await reporting.bind(ep, coordinatorEndpoint, [
      "msTemperatureMeasurement",
      "msRelativeHumidity",
    ]);
    await ep.configureReporting("msTemperatureMeasurement", [
      {
        attribute: "measuredValue",
        minimumReportInterval: 30,
        maximumReportInterval: 120,
      },
    ]);
    await ep.configureReporting("msRelativeHumidity", [
      {
        attribute: "measuredValue",
        minimumReportInterval: 30,
        maximumReportInterval: 120,
      },
    ]);
  },
};

module.exports = definition;

```

sjorge, 2023-02-18:
You can see if the PM2.5 works the same as on the starkvind.

I went to ikea to find one to add support but they're not here yet where i live.

kkossev, 2023-02-18:
> I've got one, too.

Is it already available in Ikea Germany? I can't find it online .. : (

peterkappelt, 2023-02-18:
Unfortunately, Starkvind seems to use another cluster (manuSpecificIkeaAirPurifier with id 64637), while vindstyrka seems to publish PM2.5 data on cluster 1066. Strangely, the 1066 cluster is already assigned to heimanSpecificPM25Measurement - though reading values and configuring reporting both fail with UNSUPPORTED_ATTRIBUTE.

I'll try to get around a dirigera gateway - I think capturing the real protocol is way easier than just wandering around in the dark and trial&error

@kkossev I've got mine today in a local IKEA in Germany. They didn't publish the product on their website yet, but apparently it is already available in some stores

Astrofreak85, 2023-02-18:
Today I got the last one in Ikea Dresden, they told me they got 20pcs on thursday

peterkappelt, 2023-02-18:
Okay turns out 1066 cluster isn't actually Heiman-Specific, but a general purpose PM2.5 cluster defined in the ZCL spec. So this definitely needs some patching in zigbee-herdsman (there is a manufacturer code specified for heimanSpecificPM25Measurement that prevents messages with the IKEA manufacturer code coming through).

The following definition works for me. You might need to manually click on configure and it'll probably fail, but at least the sensor is reporting temperature, humidity and PM2.5. No clue for TVOC though...
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VINDSTYRKA'],
    model: 'VINDSTYRKA',
    vendor: 'IKEA of Sweden',
    description: 'IKEA Air Quality and Humidity sensor',
    fromZigbee: [fz.temperature, fz.humidity, fz.heiman_pm25],
    toZigbee: [],
    exposes: [e.temperature(), e.humidity(), e.pm25()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const ep = device.getEndpoint(1);
        await reporting.bind(ep, coordinatorEndpoint, ["msTemperatureMeasurement", "msRelativeHumidity", "heimanSpecificPM25Measurement"]);
        await ep.configureReporting("msTemperatureMeasurement", [{attribute: "measuredValue", minimumReportInterval: 30, maximumReportInterval: 120}]);
        await ep.configureReporting("msRelativeHumidity", [{attribute: "measuredValue", minimumReportInterval: 30, maximumReportInterval: 120}]);
        await ep.configureReporting("heimanSpecificPM25Measurement", [{attribute: "measuredValue",
                        minimumReportInterval: 30, maximumReportInterval: 60, reportableChange: 1}]);
    }
};
```

peterkappelt, 2023-02-18:
@Astrofreak85 i had to do a factory reset first before I could pair it. 

Do you happen to have acess to a dirigera hub for capturing (or know anyone)? I'm also located near Dresden and would be keen to capture some data

lovwyr, 2023-02-19:
@Astrofreak85 push the Connect Button 4x to init an Reset and then its pairable
@peterkappelt when i leaved yesterday there where ~10 Units left in Stock :)

Astrofreak85, 2023-02-19:
@peterkappelt no sadly not, but if I can help anyway, just let me know

philippsandhaus, 2023-02-20:
@peterkappelt Great, this also works for me. I get the same measurements as you get. My guess is, that TVOC might not be reported over zigged at all. I tried also to use the corresponding definition for TVOC, but got no measurements:

'''
await ep.configureReporting("heimanSpecificAirQuality", [{attribute: "tvocMeasuredValue",
                        minimumReportInterval: 30, maximumReportInterval: 60, reportableChange: 1}]);
 '''
As Ikea apparently does follow the specification for PM25 I suspect they would do the same for TVOC.

peterkappelt, 2023-02-20:
@philippsandhaus Apparently, it does report TVOC data - at least that's what a [dump from the gateway](https://github.com/oleksiikutuzov/IKEA-VINDSTYRKA/blob/main/dump.json) suggests. I wasn't able to figure out the meaning of the 64xxx clusters though, they didn't respond to basic attribute read commands.

IKEA doesn't seem to fully follow the ZCL spec here either. PM2.5 reports as data type float, but per standard it should've been an integer. So we'll definitely need some manufacturer specific cluster definition. I'll try to submit a pull request till end of the week

0ip, 2023-02-20:
> Can anyone confirm this behaviour?

@peterkappelt  I tried your converter and receive PM2.5 reports as frequently as in your case.
However, I cannot even set the reporting interval. Here's the error message:
[image]

0ip, 2023-02-20:
As a side note/might be useful for later:
I read that the [SEN54 sensor from Sensirion the Vindstyrka uses](https://github.com/oleksiikutuzov/IKEA-VINDSTYRKA/blob/main/teardown.md) likely does not report a raw TVOC reading, which seems to be of little value anyway. Instead, it (only?) emits a unitless and proprietary index that can be used to detect short-term VOC changes compared to a 24 hour baseline. The baseline is 100. A value less than 100 (= less VOCs) corresponds to fresh air coming in or the use of an air purifier. Higher value = more VOCs. [See this PDF for more details about this index.](https://sensirion.com/media/documents/02232963/6294E043/Info_Note_VOC_Index.pdf)

peterkappelt, 2023-02-24:
I've created a PR with the initial support, just fyi

closed 2023-02-25 by Koenkk

Astrofreak85, 2023-03-08:
I don't know if this belongs here or not. After update to the current release I can find the Vindstyrka, but I only get the lqi readin, everything else stays at N/A

rotilho, 2023-03-09:
The temperature precision is sooooo disappointing. I hope they somehow fix it

prolife86, 2023-03-11:
> 't know if this belongs here or not. After update to the current release I can find the Vindstyrka, but I only get the lqi readin, everything else stays at N/A

Try removing it and adding it again, sounds like an incomplete interview.

prolife86, 2023-03-11:
> The temperature precision is sooooo disappointing. I hope they somehow fix it

What do you mean?

0ip, 2023-03-11:
> > The temperature precision is sooooo disappointing. I hope they somehow fix it
> 
> What do you mean?

prolife86: rotilho likely refers to the lacking number of decimals. I think it's due to the sensors intrinsic measurement inaccuracy. So it doesn't get much more precise than this in terms of hardware. [See the spec for details](https://sensirion.com/products/catalog/SEN54)

rotilho, 2023-03-12:
That's what I mean. It could at least have a 0.5 precision since the sensor has 0.45 accuracy.

sanyafifa, 2023-11-02:
> I don't know if this belongs here or not. After update to the current release I can find the Vindstyrka, but I only get the lqi readin, everything else stays at N/A

I have the same problem
How did you solve yours?

edmondss, 2024-02-07:
Hi guys, I just bought this Ikea air quality monitor. After pairing with Z2m i found out that the data in HA is NOT exactly the same as to what's showing on the ikea sensor screen. Is this normal?