Koenkk/zigbee2mqtt #16721: Battery devices leaving network when home w/o power

kizio1000, 2023-02-18:
### What happened?

When for some reason machine that is running HA goes offline ( like power outage of house or so) some battery devices (like Xiaomi RTCGQ11LM or SONOFF SNZB temperature sensors) are disappearing from network for good. Only repair brings them back and frequently they need battery to be pulled out prior pairing in order for correct pairing. 

### What did you expect to happen?

All battery devices should stay in network in case of HA machine power outage. 

### How to reproduce it (minimal and precise)

Switch off HA machine for couple of hours. 

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Sonoff ZigBee stick E

lux73, 2023-02-18:
it's quite an expected behavior with Sonoff Dongle-E because of buggy Firmware

pls search around the Forums

switch to another Coordinator is the only you could do

kirovilya, 2023-02-18:
try with latest dev/edge version z2m

wastez, 2023-02-18:
Updating to the latest edge version solved the problem for me.

wastez, 2023-02-18:
@kizio1000 
If the dev build doesn't help i have a other possibility what could help.
I had a look in the firmware change logs and there are fixes in firmware 7.2.0.0 and in 7.2.1.0 what could help.

kizio1000, 2023-02-18:
> It's save to use v7 firmware for the dongle-e?

Works much better for me

> I had a look in the firmware change logs and there are fixes in firmware 7.2.0.0 and in 7.2.1.0 what could help.

.Tried this at first but z2m/network was having stability issues so I flashed 7.1.1 which was around already for some time

kizio1000, 2023-02-18:
> try with latest dev/edge version z2m

IDK what to think, since today install of edge version I had to repair 3 devices during the day (on regular branch I did not have issues like this except power outage scenario)

wastez, 2023-02-18:
So on edge branch the devices leaves the network and you had to re-pair it? (Or come back some time ago automatically)

Do you know if the firmware can be undone to 6 if v7 is flashed?
Want to try it but I want the ability to go back. (Somewhere I read it isn't possible)

wastez, 2023-02-18:
Can you tell me what was getting better after you changed to v7?

kizio1000, 2023-02-18:
> Can you tell me what was getting better after you changed to v7?

I did 7.2 and then was able to lower it to 7.1.1. Didn't tried to go 6x

> Can you tell me what was getting better after you changed to v7?

Devices stopped leaving network (earlier I had couple of them offline every week) Hard for me to say if earlier power outage problem was present or it came up now as earlier so frequently battery devices were going offline that it was hard to distinguish if it was power outage related or not.

wastez, 2023-02-18:
Ok, can't say it how it is a week, right now i've running it for 4 days, my ikea battery devices are leaving the network every 7 to 10 minutes.
Thanks for your answers, right now everything seems to work so even if it is compelling i don't know if it should risk it to go to v7.

So with the edge version your devices were leaving your network?

kizio1000, 2023-02-18:
> So with the edge version your devices were leaving your network?

Whole last week I was running my network on latest z2m standard branch and sonoff E coordinator with 7.1.1 software, no single dropout (which was not a case on 6x firmware as I mentioned above). Last night we had 5h power outage and at the morning like 4 or 5 battery devices were gone from network.  So during normal conditions 1.30.1-1 + sonoff E with 7.1.1 were performing great, only problem is with power outage scenario. Today after I tried edge I had to repair couple battery devices already after couple of hours after switching to edge (there was no power outage during this period so for me so far 1.30.1-1 + sonoff E 7.1.1 is best combination)

maxime1992, 2023-03-21:
Up

wastez, 2023-03-21:
I can confirm the solution is Firmware 7.1.1, the original firmware seems to be very buggy.

adams-family, 2023-04-17:
> try with latest dev/edge version z2m

Is it possible when running Z2M from `hub.docker.com`?

closed 2023-05-25 by github-actions[bot]