Koenkk/zigbee2mqtt #16725: Philips Hue Wall switch module. device_mode don't work.

yoka01, 2023-02-18:
### What happened?

The module works well with the **left_press** but when I change **device_mode** from single to dual i get an error.
[image]

Same thing if I **reconfigure** or try to **delete** the device.
I forced delete one of the module, but now the module doesn't work at all. (it still works with the Philips hue app) The error : 

`Publish 'set' 'device_mode' to 'Wall Switch ' failed: 'Error: Write 0x001788010b027e88/1 genBasic({"52":{"value":2,"type":48}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4004 - 1 - 11 - 0 - 4 after 10000ms)'`

### What did you expect to happen?

I should be able to change the **device_mode** without errors or delete it without problems.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

yoka01, 2023-02-25:
It's good, I found a solution, you have to press the button which is connected to the module and at the same time modify the Device_mode (because it is in standby mode when you do not press a button).

closed 2023-02-25 by yoka01

AgileCowboy, 2024-02-20:
Best tip of the day, thanks @yoka01

latinvm, 2024-07-14:
Thanks @yoka01 pressing the buttons allowed me to switch modes :)