Koenkk/zigbee2mqtt #16726: Issue with Lidl TS0601 _TZE200_chyvmhay (Silvercrest radiator valve with thermostat)

Eistee82, 2023-02-18:
### What happened?

Controling the time shedule for the weekdays is not posible. In debug I see values recived from the zigbee device but in the ui nothing is displayed in the formular and nothing can be set. only manual mode with the setpoint is working for me. but i need automatic shedule to work.

Device definition:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7e504f2e80beaf25b19372c3ea41585db5b9c1af/devices/lidl.js#L824

### What did you expect to happen?

Set values for the schedule in webui/mqtt and transfer them to the valve.

### How to reproduce it (minimal and precise)

Pair the valve and set or receive schedule values.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

Eistee82, 2023-03-25:
problem still exists

realglotzi, 2023-03-28:
Known bug since 1.27.1

https://github.com/Koenkk/zigbee2mqtt/issues/14462

Eistee82, 2023-04-28:
Still not working

realglotzi, 2023-05-16:
seems it's working again with 1.30.4

can anyone confirm this

Eistee82, 2023-06-16:
Problem still exsists

Eistee82, 2023-10-14:
Next winter is coming and there is still no bug fix