Koenkk/zigbee2mqtt #16730: Z2M version >1.22.2 fails to start with Raspbian Buster and ConBee II

KC-inDomus, 2023-02-18:
### What happened?

Hi there,
had an old (1.13.1) Z2M instance running fine on Docker@Raspbian Buster with ConBee II with firmware 266e0700. 
Decided to make an update to latest (1.30.1): no joy, it fails to start. 

Rolled back to 1.13.1 everything works. So I tried every single later version: the last "working" is 1.22.2, starting **1.23.0 it fails to start**, so something changed starting with that version. I even tried with ConBee version 26720700 (latest) firmware, no change. 1.23.0 breaks something.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

266e0700 / 26720700

### Adapter

ConBee II

lux73, 2023-02-18:
sounds similar to https://github.com/Koenkk/zigbee2mqtt/issues/16689

KC-inDomus, 2023-02-18:
> sounds similar to https://github.com/Koenkk/zigbee2mqtt/issues/16689

I'm on docker

lux73, 2023-02-19:
https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users

`If you are running Raspbian Buster (not Bullseye!) (find out by executing grep "PRETTY_NAME" /etc/os-release) you need to install libseccomp2`

found this under "Installation/Docker" - don't know if it's related for your Installation because you don't post any error-logs

KC-inDomus, 2023-02-19:
> https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users
>
>
>
> `If you are running Raspbian Buster (not Bullseye!) (find out by executing grep "PRETTY_NAME" [...]

right, thanks. will try it

KC-inDomus, 2023-02-20:
> https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users
>
>
>
> `If you are running Raspbian Buster (not Bullseye!) (find out by executing grep "PRETTY_NAME" [...]

tried. not that.

KC-inDomus, 2023-02-21:
@Koenkk, any clue about what changed after 1.22.2 that may cause this problem?

Koenkk, 2023-02-22:
Can you provide the herdsman debug log when starting 1.30.1?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

KC-inDomus, 2023-02-22:
> Can you provide the herdsman debug log when starting 1.30.1?
> 
> 
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

sure, asap.

KC-inDomus, 2023-02-28:
Here we go with [1.22.2 log](https://pastebin.com/Nx1cUmcZ) and [1.30.0 log](https://pastebin.com/YddHAVBh).

Koenkk, 2023-03-01:
Try removing the `transmit_power` from the z2m configuration.yaml. This is not supported for the Deconz adapter.

KC-inDomus, 2023-03-05:
> Try removing the `transmit_power` from the z2m configuration.yaml. This is not supported for the Deconz adapter.

YES! Thaw was the problem. Thanks @Koenkk

closed 2023-03-05 by KC-inDomus