Koenkk/zigbee2mqtt #16731: [New device support]: MIR-TE100-TY

technoman63, 2023-02-18:
### Link

https://fr.aliexpress.com/item/1005002836127648.html?spm=a2g0o.order_list.order_list_main.47.2c515e5bTew2Hm&gatewayAdapt=glo2fra

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138aea0e3ee10","nwkAddr":65260,"manufId":4417,"manufName":"_TZE200_zl1kmjqx","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"(��+\u0011","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":72,"manufacturerName":"_TZE200_zl1kmjqx","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1676735602943,"defaultSendRequestWhen":"immediate"}

### Comments

[image]

Hello can you help me to create the definition file please ?

I send my log_debug:

[https://gist.github.com/technoman63/ecae2fb908f3617ebccde56f03547d4d](url)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_zl1kmjqx',
        },
    ],
    model: 'MIR-TE100-TY',
    vendor: 'TuYa',
    description: 'Wireless temperature & humidity detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(), e.temperature(), e.humidity()],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
```

ale1969, 2023-02-18:
Try using this code:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_zl1kmjqx',
        },
    ],
    model: 'MIR-TE-100-TY',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor with display',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.humidity(), e.linkquality(), e.battery(), 
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
		    [1, 'temperature', tuya.valueConverter.divideBy10],
                [2, 'humidity', tuya.valueConverter.divideBy10],
		    [3, 'linkquality', tuya.valueConverter.raw],
 		    [4, 'battery', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

technoman63, 2023-02-19:
Thank you, it's workink very well.

closed 2023-03-25 by technoman63