Koenkk/zigbee2mqtt #16732: Unsupported attribute apparentPower for Legrand power socket 067775/741811

JohannCR, 2023-02-18:
### What happened?

Failed to configure 'Legrand'.
OTA Update is available but does not work since configuration failed.
No value for apparentPower.
Otherwise the socket works.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20210120

### Adapter

zStack3x0

### Debug log

```
Zigbee2MQTT:error 2023-02-18 18:24:37: Failed to configure 'Legrand', attempt 1 (Error: ConfigureReporting 0x00047400000e7dde/1 haElectricalMeasurement([{"attribute":"apparentPower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.apparentPower (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:208:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:264:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))
```

Koenkk, 2023-02-19:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-19 by Koenkk

JohannCR, 2023-02-19:
Awesome it works thanks !

I got a question though : how come the devices indicated before that an update was available and now says it's running a more recent version than the one available ? (I haven't updated)
```
"installed_version": 54531584,
"latest_version": 4866563,
```
Maybe because I messed around following novisys comment in https://github.com/Koenkk/zigbee2mqtt/issues/11867 ?

JohannCR, 2023-02-21:
Have had it for like 2 years so it's not new...

mangoose002, 2023-02-25:
Unfortunately despite the update, I still get a similar error:

error 2023-02-25 09:07:07: Failed to configure 'Escalier', attempt 3 (Error: Read 0x0004740000097b1d/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.readEletricalMeasurementMultiplierDivisors (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:26:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/legrand.js:278:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13))

I tried the external converter presented in #11867 without success.

Regards.

Koenkk, 2023-02-25:
Could you check if the issue is fixed with the following external converter: 

```js
{
        zigbeeModel: [' Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: '067775/741811',
        vendor: 'Legrand',
        ota: ota.zigbeeOTA,
        description: 'Power socket with power consumption monitoring',
        fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement, fz.power_on_behavior],
        toZigbee: [tz.on_off, tz.legrand_settingEnableLedInDark, tz.legrand_identify, tz.legrand_settingEnableLedIfOn,
            tz.power_on_behavior],
        exposes: [e.switch(), e.action(['identify']), e.power(), e.power_apparent(),
            exposes.binary('led_in_dark', ea.ALL, 'ON', 'OFF').withDescription(`Enables the LED when the power socket is turned off,
                allowing to see it in the dark`),
            exposes.binary('led_if_on', ea.ALL, 'ON', 'OFF').withDescription('Enables the LED when the device is turned on'),
            e.power_on_behavior()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
            await reporting.activePower(endpoint);
            try {
                await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
                await reporting.apparentPower(endpoint);
            } catch (e) {
                // Some version/firmware don't seem to support this.
                // https://github.com/Koenkk/zigbee2mqtt/issues/16732
            }
        },
    }
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed. Make sure that the reported `power` value makes sense.

JohannCR, 2023-02-25:
If I'm not mistaken it's next to the configuration.yaml inside the zigbee2mqtt folder and not the HA config folder, right ?
I wasn't sure either from the other thread...
Just saying for those who need to try that.

Koenkk, 2023-02-26:
In the Zigbee2MQTT folder indeed.