Koenkk/zigbee2mqtt #16736: Sonoff Dongle Plus keeps dropping out after 20221226 upgrade

Morphy99, 2023-02-18:
### What happened?

My Sonoff coordinator keeps locking up and nothing responds until I restart Z2M. This happens intermittently and I can't see anything in the logs apart from ping failures from availability checks (`failed (SRSP - AF - dataRequest after 6000ms)`). This happens randomly, sometimes 3-4 a day, some days never, some days once.

This appears to have happened since upgrading to Coordinator revision 20221226 as before this I had I've tried reflashing firmware, moving the coordinator physically, it's already on a 0.5m USB3 extension lead from the NUC. My WiFi AP's are on channels 1 and 13 and my Zigbee channel is 15. There is no neighbouring AP's above -70 so I don't think it's WiFi related.

My Z2M network comes from a CC2530 and Raspberry Pi 4 which I upgraded to a NUC and a Sonoff Dongle Plus a few months ago. I have a varied manufacturer network with 37 end devices and 18 routers. After setting the secondary MAC the upgrade worked with a few issues here and there and a few things needing re-pairing. I had a lot of issues with Xiaomi contact sensors dropping off and a couple of IKEA bulbs doing very weird things which I've put down to them being faulty.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle Plus

Koenkk, 2023-02-19:
Does downgrading back to `20221226` fix the issue?

Morphy99, 2023-02-19:
I'll give that a go, thanks. I didn't know if downgrading would cause other issues or not. 
I also read about disabling home assistant USB discovery, although that's going to be slightly harder to implement as I use default config and there is no easy way of disabling one component. I don't think that's related though, as it seemed to work fine before the coordinator upgrade but it could well be a HA update has done something...

Morphy99, 2023-02-19:
So it seems after downgrading this morning it's been fine all day until it just reoccurred. So maybe it isn't the firmware.

Morphy99, 2023-02-24:
Still trying to get to the bottom of this. Things I've tried:

- Changed USB cable extension for a 1m USB 2.0 shielded one (I was using a 1m USB 3.0)
- Disabled USB auto-discovery in HA by removing `default_config:` in configuration.yaml and only specifying needed integrations.
- moving coordinator around but in the same general area.
- downgrade coordinator firmware to `20221226`

I did have something interesting occur the other day where restarting Z2M did not resolve the issue for very long, it kept reoccurring after only a few minutes of it restarting. This kept happening until I restarted the host. So I'm wondering if this was something to do with the OS or maybe it was the power cycle of the coordinator. Either way it hasn't occurred again since so I can't try just removing the coordinator instead of restarting. 

Does anyone have any suggestions how I can debug this further? 

I'm going to try an OS upgrade to 9.5 as I'm on 9.4 but other than that, I'm stuck on what to try next.

Morphy99, 2023-02-24:
There is a neighbouring network on channel 3 but the strength is very low, -81dbm. Could this be the cause?
[image]

closed 2023-03-19 by Morphy99