Koenkk/zigbee2mqtt #16737: TZ3210_onejz0gt / TS0504B - cannon change color via HA

fabiobaldoibs, 2023-02-18:
### What happened?

After the update to version 1.30.0-x, I cannot change the rgb color on HA. I can change the color by using the exposes feature inside the ZigBee2MQTT module. I returned the backup to 1.29.2-1 and now it is working. 

### What did you expect to happen?

Change the color

### How to reproduce it (minimal and precise)

Inside the lovelace, find the device and click the menu to open the color settings. Click to change de color - nothing happened. 

### Zigbee2MQTT version

1.29.2-1

### Adapter firmware version

20220219

### Adapter

sonoff zibnee 3.0

marciej123, 2023-02-23:
Same Issue over here, didn't test the older version of Z2M.
I have the Slaesh's CC2652RB stick.

paulbrause, 2023-02-23:
Same issue here as well. Older Version not tested.
My Coordinator is Phoscon ConBee II.

LeszekBi, 2023-03-02:
Same for me. 
In debug mode I noticed one difference:
send from Z2M -> Expose (it's ok)
Received MQTT message on 'zigbee2mqtt/entity_name/set' with data '{"color":{"hex":"#86007D"}}'
send from Lovelace (not ok)
Received MQTT message on 'zigbee2mqtt/entity_name/set' with data '{"state":"ON","color":{"h":360.0,"s":85.71428571428571}}'

fabiobaldoibs, 2023-03-05:
@LeszekBi
Where can I find the "color_mode" configuration? or is a yaml configuration?

LeszekBi, 2023-03-05:
@fabiobaldoibs I can see it in z2m -> MyDevice -> Status -> "color_mode": "hs".
For now I don't know how to change color_mode by config but I noticed that it's changing when I use z2m interface to change color and the same on lovelace on different cards. (card decide which color_mode to use and change it).

paulbrause, 2023-03-14:
Z2M -> MyDevice -> Status show me this:
```
{
    "brightness": 254,
    "color": {
        "h": 210,
        "hue": 210,
        "s": 85.71428571428571,
        "saturation": 85.71428571428571,
        "x": 0.1873,
        "y": 0.2552
    },
    "color_mode": "hs",
    "linkquality": 252,
    "state": "ON"
}
```

When i change the color via Z2M (works fine, no matter if i change color_hs or color_xy):
```
2023-03-14 18:32:45 Received MQTT message on 'zigbee2mqtt/LEDStreifen_Bett_Schlafzimmer/set' with data '{"color":{"hex":"#0000F9"}}'
2023-03-14 18:32:45 Skipping state because of Home Assistant
2023-03-14 18:32:45 Publishing 'set' 'color' to 'LEDStreifen_Bett_Schlafzimmer'
2023-03-14 18:32:45 Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0xa4c138d4881f34f7!
```

When i try to change the color via HA Lovelace:
```
2023-03-14 18:35:01 Received MQTT message on 'zigbee2mqtt/LEDStreifen_Bett_Schlafzimmer/set' with data '{"state":"ON","color":{"h":120.0,"s":85.71428571428571}}'
2023-03-14 18:35:01 Skipping state because of Home Assistant
2023-03-14 18:35:01 Publishing 'set' 'color' to 'LEDStreifen_Bett_Schlafzimmer'
2023-03-14 18:35:01 Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0xa4c138d4881f34f7!
```

Where is the problem now? Z2M or Lovelace?

And any idea how to fix this?

LeszekBi, 2023-03-14:
I think Lovelace use color_mode from entity attribute. You can try to switch between modes by calling service: 
Developer tools -> Services -> light.turn_on
There you have nice GUI to set proper options.

paulbrause, 2023-03-14:
I tried to switch to another color mode, but without success:

Via Z2M (working):
```
2023-03-15 00:53:34 MQTT publish: topic 'zigbee2mqtt/LEDStreifen_Bett_Schlafzimmer', payload '{"brightness":251,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","linkquality":255,"state":"ON"}'
```

Via Developer tools -> Services (not working):
```
2023-03-15 00:55:34 MQTT publish: topic 'zigbee2mqtt/LEDStreifen_Bett_Schlafzimmer', payload '{"brightness":251,"color":{"h":39.6,"hue":39.6,"s":78.431,"saturation":78.431,"x":0.4362,"y":0.4211},"color_mode":"hs","linkquality":255,"state":"ON"}'
```

So it is still using "color_mode: hs" instead of "color_mode: xy".

fabiobaldoibs, 2023-04-18:
Did anyone try to update to the latest version?

closed 2023-05-27 by github-actions[bot]

LasseTheDude, 2023-12-11:
Something new here? Very frustrating. I go from ZHA to Z2M and discovered this problem now :(

adriweb, 2024-03-22:
Same issue with homebridge-z2m, which uses `hs` colorMode, which doesn't work (everything is fine via z2m itself, which uses `xy`)