Koenkk/zigbee2mqtt #16747: Coordinator sends endless open/close commands upon device reconnect.

dantidote, 2023-02-19:
### What happened?

First of all, apologies if this isn't the right place to report this.  Can't really tell if this is a z2m, TI, or IKEA issue.

I have a bunch of Ikea Tredansen blinds in my house.  When the battery dies, I recharge overnight and reconnect the battery.  A few seconds later, it seems to move itself up and down endlessly. It stops when I disconnect the zigbee stick.  I finally bought a sniffer, and it looks like these commands are being sent by the coordinator, but I don't see any relevant log entries in z2m.

Video:
https://youtu.be/AXUuh9g0k_c

Packet Captures:
[pcap.zip](https://github.com/Koenkk/zigbee2mqtt/files/10777170/pcap.zip)

### What did you expect to happen?

Coordinator shouldn't send random open/close commands.

### How to reproduce it (minimal and precise)

1. battery dies
2. recharge overnight
3. connect battery

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

[0xb41b.log](https://github.com/Koenkk/zigbee2mqtt/files/10777225/0xb41b.log)
[0xd17f.log](https://github.com/Koenkk/zigbee2mqtt/files/10777226/0xd17f.log)

karaminder, 2023-03-02:
Wanted to add my Fyrtur Blinds are behaving the same way in your video.

I too have a dense zigbee network.

My coordinator is the: SMLight SLZB-06.

Noticed in your logs that we may have something else in common… Inovelli blue light switches? Just an idea. I’m out of my depth here.

Kimovitzh, 2023-03-13:
Same for me!

It's been going on for more than tree weeks for me, and it seems to stop when I do kill Z2M with 
`sudo systemctl stop zigbee2mqtt`
OR, if I removes the curtain from Z2M and reconnects.

This is a big issue to the point where I either throw the curtains out!

Please fix 😩

closed 2023-04-20 by github-actions[bot]