Koenkk/zigbee2mqtt #16748: [New device support]: Need help adding Lerlink (Tuya) Fan Switch (5 speed)

isakharov, 2023-02-19:
### Link

https://aliexpress.ru/item/1005004705097813.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138d758a8441b","nwkAddr":8058,"manufId":4417,"manufName":"_TZE200_r32ctezx","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE200_r32ctezx","zclVersion":3,"appVersion":70,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1676822088873,"defaultSendRequestWhen":"immediate"}

### Comments

Hey! I have nice Zigbee Fan Switch with 5 speed [https://aliexpress.ru/item/1005004705097813.html](url)
It looks like a dimmer from this issue [https://github.com/Koenkk/zigbee2mqtt/issues/4413](url) but it isn't dimmer and dimmer converter for this device doesnt work at all.

Using TuYA gateway I've got all tuyaDatapoints using this page [https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html](url):
fan switch - Code 1
fan countdown - Code 2
fan speed - Code 3
power on state setting - Code 11

and all Standart Instruction Set from Device Debugging Mode:
switch_fan	Boolean	
"{true,false}"
countdown_fan	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 43200,
  "scale": 0,
  "step": 1
}
fan_speed_enum	Enum	
{
  "range": [
    "level_1",
    "level_2",
    "level_3",
    "level_4",
    "level_5"
  ]
}
relay_status	Enum	
{
  "range": [
    "off",
    "on"
  ]
}

I made converter for this. 
It can switch on/off and get speed after physically changing on devise.
Also it looks like that power on state could get correctly.

Unfortunately changing of speed, power on state and countdown from z2m doesn't working.
I've tried converters from brightness, switch and others but it doesn't help.
I have no idea what to do to change parameters of this fan switch.
Need your help. Thank you!

P.S. I have tuya gateway and CC2531 sniffer with PACKET-SNIFFER firmware

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'Lerlink Fan Switch',
    vendor: '_TZE200_r32ctezx',
    description: 'Lerlink Fan Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
	configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.switch(), 
	          tuya.exposes.countdown().withValueMin(0).withValueMax(43200).withUnit('seconds').withDescription('Max on time in seconds'),
			       exposes.numeric('fan_speed', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1).withDescription('Fan speed'),
				   e.power_on_behavior().withDescription('Power-on state'),
	],

	meta: {
		tuyaDatapoints: [                
                [1, 'state', tuya.valueConverter.onOff], //Boolean
				[2, 'countdown', tuya.valueConverter.countdown], //Integer
                [3, 'fan_speed', tuya.valueConverterBasic.lookup({'1': 0, '2': 1, '3': 2, '4': 3, '5': 4})], //enum			
				[11, 'power_on_behavior', tuya.valueConverter.powerOnBehavior], //enum
	        ],
		},
};

module.exports = definition;
```

isakharov, 2023-02-24:
So, thanks community [zigbeer](https://t.me/zigbeer) for help.
here the right converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'Lerlink Fan Switch',
    vendor: '_TZE200_r32ctezx',
    description: 'Lerlink Fan Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
	configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.switch(), 
	          tuya.exposes.countdown().withValueMin(0).withValueMax(43200).withUnit('seconds').withDescription('Max on time in seconds'),
			       exposes.numeric('brightness', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1).withDescription('Fan speed'),
				   exposes.enum('power_on_behavior', ea.STATE_SET, ['off', 'on']).withDescription('Relay state'),
	],

	meta: {
		tuyaDatapoints: [                
                [1, 'state', tuya.valueConverter.onOff], //Boolean
				[2, 'countdown', tuya.valueConverter.countdown], //Integer
                [3, 'brightness', tuya.valueConverterBasic.lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})], //enum				
				[11, 'power_on_behavior', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on': tuya.enum(1)})], //enum
	        ],
		},
};

module.exports = definition;
```

isakharov, 2023-02-24:
Create PR for it [Add support for Lerkink Fan Switch](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5522)

d0xt0p, 2024-09-13:
Good afternoon. I received a similar device, but when connecting I ran into the problem "not supported". Configured via the converter as described above. In z2m, it is displayed as in a picture. Product link:https://aliexpress.ru/item/1005007085179328.html?sku_id=12000039348389078
[image]