Koenkk/zigbee2mqtt #16752: Aqara roller shade companion E1 / unknown key

delinend, 2023-02-19:
### What happened?

When Aqara roller shade companion E1 reports status, there are 2 "unknown key" lines in the debug message. Shade works fine.

Have tryed to Force forget/remove the device in Zigbee2mqtt, but after new join, the message is coming back again.

Aqara roller shade companion E1 firmware: ZNJLBL01LM (latest).

Using stas-demydiuk/domoticz-zigbee2mqtt-plugin and I have also trued to remove device in Domoticz, before rejoining.

### What did you expect to happen?

No "unknown key" in debug.

### How to reproduce it (minimal and precise)

Constant debug errors.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

Sonoff zigbee 3.0 usb dongle plus

### Debug log

Zigbee2MQTT:debug 2023-02-19 20:59:05: Received Zigbee message from 'Roller - Bryggers1', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,21,5,33,1,0,9,33,0,1,10,33,0,0,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,0,101,32,100,102,32,192,103,32,0,105,32,2,106,33,230,31,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-19 20:59:05: lumi.curtain.acn002: Processed buffer into data {"3":21,"5":1,"9":256,"10":0,"11":0,"12":1,"13":3870,"17":1,"100":0,"101":100,"102":192,"103":0,"105":2,"106":8166,"107":0}
**Zigbee2MQTT:debug 2023-02-19 20:59:05: lumi.curtain.acn002: unknown key 13 with value 3870
Zigbee2MQTT:debug 2023-02-19 20:59:05: lumi.curtain.acn002: unknown key 17 with value 1**
Zigbee2MQTT:debug 2023-02-19 20:59:05: lumi.curtain.acn002: Processed data into payload {"device_temperature":21,"power_outage_count":0,"state":"OFF","battery":100}
Zigbee2MQTT:debug 2023-02-19 20:59:05: lumi.curtain.acn002: Processed data into payload {"device_temperature":21,"power_outage_count":0,"state":"OFF","battery":100}
Zigbee2MQTT:info  2023-02-19 20:59:05: MQTT publish: topic 'zigbee2mqtt/Roller - Bryggers1', payload '{"battery":100,"device_temperature":21,"linkquality":105,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF"}'

Gr4ffy, 2023-02-20:
Maybe it's related to firmware? 
Mine have older firmware and there is no problem.

`{
    "battery": 64,
    "charging_status": false,
    "device_temperature": 21,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-02-20T13:18:54+01:00",
    "linkquality": 255,
    "motor_state": "pause",
    "position": 100,
    "power_outage_count": 0,
    "running": false,
    "state": "OFF",
    "temperature": 23,
    "update": {
        "installed_version": 3611,
        "latest_version": 3870,
        "state": "available"
    },
    "update_available": true
}`

delinend, 2023-02-20:
Yes. You got firmware version 3611 and have updated to 3870.

Hope someone can fix it.

mamrai1, 2023-02-23:
MQTT publish: topic 'zigbee2mqtt/Right Kitchen Blind', payload '{"battery":86,"charging_status":false,"device_temperature":15,"last_seen":"2023-02-23T07:37:45+02:00","linkquality":128,"motor_state":"pause","position":89,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":null}'

mamrai1, 2023-02-23:
i have also seen that error:

2023-02-23 07:39:54Received Zigbee message from 'Left Kitchen Blind', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,16,5,33,1,0,9,33,0,1,10,33,4,75,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,40,101,32,87,102,32,128,103,32,31,105,32,2,106,33,226,30,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2023-02-23 07:39:54lumi.curtain.acn002: Processed buffer into data {"3":16,"5":1,"9":256,"10":19204,"11":0,"12":1,"13":3870,"17":1,"100":40,"101":87,"102":128,"103":31,"105":2,"106":7906,"107":0}
Debug 2023-02-23 07:39:54lumi.curtain.acn002: unknown key 13 with value 3870
Debug 2023-02-23 07:39:54lumi.curtain.acn002: unknown key 17 with value 1
Debug 2023-02-23 07:39:54lumi.curtain.acn002: Processed data into payload {"device_temperature":16,"power_outage_count":0,"state":"OFF","battery":87}
Debug 2023-02-23 07:39:54lumi.curtain.acn002: Processed data into payload {"device_temperature":16,"power_outage_count":0,"state":"OFF","battery":87}

mamrai1, 2023-02-23:
unknown key 13 with value 3870 is probably firmware version on device...

delinend, 2023-02-23:
Yes. Key 13 is FW version. Don't know what key 17 with value 1 is.

closed 2023-04-02 by github-actions[bot]

mamrai1, 2023-04-02:
Can we fix this?