Koenkk/zigbee2mqtt #16753: no converter available for 'light indicator level' ("") (ZBDongle-E)

Kitchigo, 2023-02-19:
### What happened?

I have this message with a SonOff  ZBDongle-E with router firmware : ``no converter available for 'light indicator level' ("")``

### How to reproduce it (minimal and precise)

Just with a router firmware

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20210708

### Adapter

ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10777949/log.txt)

Kitchigo, 2023-03-11:
I'm now with Zigbee2MQTT 1.30.2 and firmware zStack3x0 20221226 , but same error...

closed 2023-04-19 by github-actions[bot]

HoM3r17, 2023-07-09:
I'm using a ZB Dongle E as a router since a few month, and have the same issue, it doesn't seem to affect anything badly, but it shows in the logs every minute...