Koenkk/zigbee2mqtt #16755: Danfoss 014G2461 can't set temperature/parameters

ipeacocks, 2023-02-20:
### What happened?

I have 3 Danfoss Ally (014G2461) TRVs. They are really cool but from time these TRVs become unresponsive (all of them, even tried additional one): they are online, send data to coordinator (Zigbee2MQTT/Home Assistant) but it's not possible to send data to them (set temp or change any parameter from Zigbee2MQTT web panel).  As a failure message Zigbee2MQTT says:
```
Publish 'set' 'occupied_heating_setpoint' to 'Guest Thermostat' failed: 'Error: Command 0x847127fffef42d30/1 hvacThermostat.danfossSetpointCommand({"setpointType":1,"setpoint":2100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4678,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37182 - 1 - 13 - 513 - 11 after 10000ms)'
```

That how it looks from Z2M debug console. TRV just doesn't response at all:
```
2023-02-20 01:53:17Received MQTT message on 'zigbee2mqtt/Guest Thermostat/set' with data '{"occupied_heating_setpoint":16.5}'
2023-02-20 01:53:17Publishing 'set' 'occupied_heating_setpoint' to 'Guest Thermostat'

2023-02-20 01:53:40Error: Command 0x847127fffef42d30/1 hvacThermostat.danfossSetpointCommand({"setpointType":1,"setpoint":1650}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4678,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37182 - 1 - 32 - 513 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
2023-02-20 01:53:45Received Zigbee message from 'Guest Lamp', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784918,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0
```

In general that occurs when router which was used by TRV goes down. It looks like Danfoss Ally can't select new router after this.

From Z2M Map that looks as follow https://i.imgur.com/r0CudS7.png

### What did you expect to happen?

I expect working `set` command after ZigBee router change in chain.

### How to reproduce it (minimal and precise)

Hard to say. Try to turn on/off router in chain with TRV or turn off all routers except coordinator for a few mins.

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20211217

### Adapter

Electrolama zig-a-zig-ah! (zzh!) CC2652

lux73, 2023-02-20:
first try to flash your Coordinator with the latest 20221226 Firmware. This solves a similar Problem with one oft my four Danfoss TRV

Its allways bad for Battery powered Devices if Routers go offline/online/offline and so in

I cant confirm that the Allys will not pick another Router
after installing an additional Hue GU10 Spot my Ally had pick up these Router after 24h as best Route

ipeacocks, 2023-02-20:
Actually I've found this problem as separate issue https://github.com/Koenkk/zigbee2mqtt/issues/13478
I have exactly the same.

> Its always bad for Battery powered Devices if Routers go offline/online/offline and so in

Yes, but I think it still should find working router.

closed 2023-03-30 by github-actions[bot]