Koenkk/zigbee2mqtt #16758: Interruptor inalámbrico Legrand 067773

pikatoste81, 2023-02-20:
### What happened?

I have migrated all my Legrand devices to HA using ZigBee2MQTT.  With Legrand's gateway everything worked correctly. Now wireless commands repeat the action on some occasions. 

For example, I press the command to turn on a light and a second later, turn it off. It also happens in reverse. Not always but many more times than desirable. I even are I even consider using Legrand's gateway.

All my commands have version 003C or 003F and my version of ZigBee2MQTT in Home Assistant is 1.30.1-1.

I have seen some threads with similar problems. Finally they fixed them but the versions of the devices were inferior.

### What did you expect to happen?

I hoped that with a pulsation, a single action will be executed. Not two.

### How to reproduce it (minimal and precise)

Updating the device to the latest version through the Legrand gateway and then use it to Home Assistant with ZigBee2MQTT.

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Dongle

marektryner, 2023-04-06:
@pikatoste81 same issue here. I decided to try connect the Legrand wireless switch with ZHA integration in Home Assistant and it works much better.

There is another thing which works a bit better in Zigbee2MQTT - doesen't matter where you push the switch, there is a change of state of the light. Fine thing!! But the ON button is down and OFF is on the top. Strange thing, but it is same if you connect the remote to Legrand Gateway.

In ZHA the switch behave a bit different. Top of the button switch is ON state and down button is OFF state. Also the wireless remote must be upside down compare to Zigbee2MQTT for correct dimming positions. But from my opinion correctly as the ON mark on the button is UP and OFF mark is DOWN.

I hope that this problem will be repaired soon as is very anoying and i really miss that fine thing - change the state of the light regardless of where I press the button!

Also the state of the battery (which is possible to view in Legrand Home + Control app) would be perfect!

Those two things and those Legrand switches will be the perfect thing to control my IKEA / HUE bulbs trought HA.

pikatoste81, 2023-04-09:
It seems that it is not a device firmware problem. 

I recently linked two new wireless commands (without updating) from Legrand to HAOS. I also migrated two wired switches from the Legrand gateway to HOAS.One of the new wireless commands with an old firmware version (002) has the same problem. On the other hand, one of the wired switches stopped communicating its status to HAOS and even stopped working. It wouldn't turn on or off. I removed it from the device list and paired it several times but it didn't fix the problem. Neither lower and raise the automatic of the electricity box. One of my blind switches also stopped working. All of them work correctly when linked to the Legrand gateway.

It seems that the problem occurs when reaching a certain number of Legrand devices linked via zigbee to HA.

I also hope that this problem will be solved soon because it is extremely annoying. I have configured the devices dozens of times. And the Netatmo or Homekit integration is not working correctly. Otherwise, I would use these integrations and forget about zigbee2mqtt.

marektryner, 2023-05-10:
The problem remains

pikatoste81, 2023-05-24:
> Hay otra cosa que funciona un poco mejor en Zigbee2MQTT: no importa dónde presione el interruptor, hay un cambio de estado de la luz. Cosa fina!! Pero el botón de ENCENDIDO está abajo y el de APAGADO arriba. Cosa extraña, pero es lo mismo si conectas el mando a Legrand Gateway.
> 
Is it possible to change the operation of the physical buttons of the wired switch using ZHA? I am testing with a switch and I would like the toggle function to be activated by pressing any zone.

marektryner, 2023-05-25:
@pikatoste81 sorry i don't remember that! I think in the ZHA was ON action and OFF action and not toggle. So there was necessary to push the switch on correct zone. 

But after so much pain and issues with the wired switch binded with 2 wireless switches (also Legrand) i moved to wireless Legrand switches only - each one is binded with the group of smart ZiggBee bulbs - and now, everything works perfectly!

Battery status is fine, connection stable, toggle action works perfectly (so there is possible to push the swich anywhere you want and the opposite action is performed on the light). There is also possible to dimm light (group of bulbs) correctly. 

As the wireless switch is directly binded with bulb groups then it is not dependent on the running Home Assistant / Z2M and works perfectly even those two thing are shut down.

I now recommend this solution, as it is not a wired but wireless only, so there is a necessary to change the battery on the switches each 1-2 years i think.

amgag, 2023-05-31:
I have the exact same problems for months now. My girlfriend is going crazy :)
The messages are sent twice, with a ~2 second delay, when pressing a button about 80% of the time.
```
Info 2023-05-31 11:38:34MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9', payload '{"action":"toggle","battery":80,"linkquality":94,"voltage":2900}'
Info 2023-05-31 11:38:34MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9', payload '{"action":"","battery":80,"linkquality":94,"voltage":2900}'
Info 2023-05-31 11:38:34MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9/action', payload 'toggle'
Info 2023-05-31 11:38:36MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Inter Couloir WC I9","ieee_address":"0x0004740000aaa866"},"type":"device_announce"}'
Info 2023-05-31 11:38:36MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9', payload '{"action":"toggle","battery":80,"linkquality":94,"voltage":2900}'
Info 2023-05-31 11:38:36MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9', payload '{"action":"","battery":80,"linkquality":94,"voltage":2900}'
Info 2023-05-31 11:38:36MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9/action', payload 'toggle'
Info 2023-05-31 11:38:36MQTT publish: topic 'zigbee2mqtt/Inter Couloir WC I9', payload '{"action":null,"battery":80,"linkquality":91,"voltage":2900}'
```

I tried updating the firmware of the switches to the very latest version 0x3f. The problem is still occurring.
It is also frustrating that the payload only contains the action "toggle" if I press the UP or DOWN button.

pikatoste81, 2023-07-01:
The problem persists. I am clear that the firmware version is not decisive, although it may occur less frequently in the old versions. 

When I have had few Legrand devices, be they commands or wired switches, everything has worked perfectly. As soon as I add some more device, the problems return.

Devices connected by ZHA do not have duplication problems of actions but have less functionalities.

amgag, 2023-08-02:
The problem is still occurring and prevents from properly using the Legrand
switches.

Diaoul, 2023-08-31:
What is the best way to help solve this?

closed 2023-10-09 by github-actions[bot]

oywino, 2024-09-26:
To all you fellows out there; Why are you not using the standard Legrand Gateway with all your Legrand devices?
Then, if you add the Netatmo integration to Home Assistant, all thse Legrand devices will automatically show up in HA and you can control them with any issues. I do. I have roller shutters, lights, switches, dimmers etc. from Legrand and they all work perfectly.
I never saw the advantage of trying to pair these devices directly with Z2M or ZHA?

Diaoul, 2024-09-26:
> lights

So you don't use Z2M basically for lights is what you are saying? I don't want to depend on some cloud for my light automation to turn on or off personally (motion lights).
And this adds noticeable delays.

oywino, 2024-09-26:
The Legrand Ecosystem as such is not cloud-dependant. But the integration with Home Assistant is either Netatmo cloud or Apple HomeKit cloud.
This means you'll always be able to turn your lights on/off/dim using your wall switches, even if internet is down. 
I am not aware of any non-cloud based integration from Legrand to HA, but it would be nice to have.

oywino, 2024-09-27:
> > Yes but no automation would work, like motion lights. And even with internet, it would add noticeable delay.
> It is correct of course that automations would fail for a cloud based integration which happens to be down, but as long as it works - I haven't noticed any delays.

> > I am not aware of any non-cloud based integration from Legrand to HA, but it would be nice to have.
> 
> Z2M 😆
> Well, if you pair all your Legrand devices with Z2M, then I assume they will no longer respond to direct commands from Legrand wireless wall switches (lights, roller shutters etc)?  And you will no longer be ble to control them using the H+S App (in case HA is down).  So I guess there are advantages and disadvantages wherever you turn.

Diaoul, 2024-09-27:
> I assume they will no longer respond to direct commands from Legrand wireless wall switches (lights, roller shutters etc)?

Why so? They will, I [bind](https://www.zigbee2mqtt.io/guide/usage/binding.html) my light switches to the lights in Z2M. They don't depend on HA at all.

oywino, 2024-09-27:
I see, well perhaps that's possible. I've never tried that. I was under the impression that Legrand only partly complied with the ZigBee standard. But if you've done this successfully I'll take your word for it.