Koenkk/zigbee2mqtt #16768: [New device support]: FUT035Z+

MotoMoto42, 2023-02-21:
### Link

https://miboxer.com/en/product/2-in-1-led-controller-zigbee-3-0-2-4g

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c138f2561e7d4f","nwkAddr":12086,"manufId":4417,"manufName":"_TZB210_lmqquxus","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1" :{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502B","manufacturerName":"_TZB210_lmqquxus","power Source":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOn Off":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[], "meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1676969769316,"defaultSendRequestWhen":"immediate"}

### Comments

Good day!

I try to pairing FUT035Z+ to z2m, but device detected as TUYA Device:
Model Zigbee:               TS0502B
Manufacture Zigbee:      _TZB210_lmqquxus

Next, I manually added to miboxer.js for device FUT035Z my id:
{modelID: 'TS0502B', manufacturerName: '_TZB210_lmqquxus'}

and reboot Z2M. After this, my FUT035Z+ detect as FUT035Z, and all working good. But don't work parameter "power_on_behavior" - I get Error "Unsupported_Attribute" message

Please, add this device to z2m 

### Supported color modes

Dual White LED Controller

closed 2023-04-01 by github-actions[bot]

lorobis, 2023-04-17:
Have the same problem, how can I fix?

roytomeij, 2023-05-18:
Before I buy a FUT035Z+: does anyone know whether this device is fully supported?

lorobis, 2023-05-18:
It's work decently, but some futures not working, I use it with an external converter, not yet fully supported

wardwolfram, 2023-08-18:
I am also looking into this controller. Would vote for Z2M support!

MadWalnut, 2023-10-18:
Any chance to get this issue re-opened @Koenkk? I'm also looking for support of this device.

Koenkk, 2023-10-18:
- Fixed the device detection
- Given that you get an `Unsupported_Attribute` error when trying to set it, it means the device does not support this.

MadWalnut, 2023-10-18:
@Koenkk Thank you for that super fast commit!

stefan-oskamp, 2024-01-04:
I have two of these devices. Both have V2.1 printed on them. One was detected okay the other was detected as a Tuya device with RGB color (not dual-white).

The one that was not detected correctly, has a model number of TS050**3**B. So I duplicated the definition for TS0502B in tuya.js and changed the model and zigbeeModel to "TS0503B". That is probably not the right way, but it fixed it for me.

I am using z2m version 1.35.0.

The generated external definition in the dev console is:
```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0502B'],
    model: 'TS0502B',
    vendor: '_TZB210_lmqquxus',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[153,500]}})],
};

module.exports = definition;
```
Would be nice if this can be added in a future release.

titou4307, 2024-09-13:
Looking for to replace my controler Miboxer FUT035W+ (Wifi) by the version FUT035Z+ (zigbee 3.0)

Finally does this controller FUT035Z+ is fully supported by Z2M ?

Koenkk, 2024-09-14:
@stefan-oskamp thanks, integrated the fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

bzumik1, 2024-10-02:
> It's work decently, but some futures not working, I use it with an external converter, not yet fully supported 

Could you please share the link? I am missing the startup color setting.

bzumik1, 2024-10-02:
> - Fixed the device detection
> 
> - Given that you get an `Unsupported_Attribute` error when trying to set it, it means the device does not support this.
> 
> 
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

This device should support power_on_behavior functionality. I have double checked it with manufacturer.

doblum, 2024-11-23:
I have now the V3.0 edition and is it recognized as TS0502B.
The Manufacturer Number ist _TZB210_ue01a0s2

Could this also be integrated ?
Thank you and best regards
Dominik

Kapteenivamma, 2024-12-12:
Same problem here. They worked very well for 6months and they were  recognized correctly as FUT035Z+.

In some point all of the 3pcs of FUT035Z+ lost abilty for light transition and also some times they only light up with minimum brightness with motion detector. 

I started to investigate and find out that they are recognized as TS0502B.

dawiinci, 2024-12-23:
Please fix this, as my lights are constantly forgetting the last set color. I assume this is related.