Koenkk/zigbee2mqtt #16769: SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P automatic flashing instructions are broken

Barsonax, 2023-02-21:
### What happened?

The pdf asks you to Download, unzip and execute the file however its not clear which file this is. I suspect a link is broken here:
[image]

I tested in Brave, Chrome and Edge and all have the same result. Word does show a uartlog.py, which is the script I need I suspect, but I cannot open it

I tried to create a issue at https://github.com/Koenkk/zigbee2mqtt.io but this seems to be disabled there so iam assuming this repo is the correct spot.

### Zigbee2MQTT version

not applicable

### Adapter firmware version

not applicable

### Adapter

not applicable

lux73, 2023-02-21:
https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_via_cc2538-bsl.html

dont know from where the mentioned PDF is  - but if you want to flash your Coordinator all is described in the postet link 😉

Barsonax, 2023-02-21:
https://www.zigbee2mqtt.io/guide/adapters/

It's here if you go to that sonof dongle

closed 2023-04-01 by github-actions[bot]