Koenkk/zigbee2mqtt #16773: Occasionally occupancy stays ON indefinitely (TS0202 / _TZ3000_6ygjfyll)

gu33mis, 2023-02-21:
### What happened?

I recently purchased 5x TuYa IH012-RT01 motion sensors which are recognised by my zigbee2mqtt instance as _TZ3000_6ygjfyll model TS0202. Everything seems to (almost) work fine, but occasionally occupancy on random sensors would stay ON all the time. The only way to force it to go OFF is to trigger the motion and after the timeout set for the sensor (keep_time) it will then go OFF properly.

It seems to be random and all 5 sensors are experiencing it. I have approx. 40 other devices inc. other zigbee PIR's and I never had that problem before.

### What did you expect to happen?

I would expect the 'occupancy' sensor to go OFF after configured 'keep_time'

### How to reproduce it (minimal and precise)

Random occurrence., so might be difficult to reproduce it. Once the motion is detected ("occupancy" is ON), sometimes the sensor will change to OFF after 'keep_time', but sometimes it will stay set to "ON" until motion is detected again.

### Zigbee2MQTT version

1.30.1 commit: eb878d3

### Adapter firmware version

0x26780700

### Adapter

ConBee2

closed 2023-04-01 by github-actions[bot]

pvanhouten, 2023-04-12:
Bump. I'm having the same issue. It seems to happen to me when there's an extended period motion beyond the keep time.  If I move around in front of the sensor for several minutes, it continues to see occupancy for that entire period and fails to detect when I've left. It's almost like it falls asleep while in a period of active motion detection.

ggggh, 2024-11-14:
Same here, still happening on latest version