Koenkk/zigbee2mqtt #16776: Delay in activation Avatto TS0006

rafareusch, 2023-02-22:
### What happened?

During normal usage, ( TS0006 / _TZ3000_jyupj3fw ) sometimes have a big delay responding to commands via Z2M, works fine physically. 

I have saw similar issue with the (TS0003 / _TZ3000_eqsair32) but much, much less common.

Comments:
- I have two different HA/Z2M instances that use this switch, all of them have this issue. Other two instances I have use another 6 gang switch and works fine with similar HA/Z2M Configuration (almost exactly the same, restored via backup copy)
- Using via TUYA integration with a Zemismart Zigbee Gateway(ethernet) the switch doesn't appear to present this behavior.
-I believe to be the first user of this Avatto switch, which support was included here [#4769](https://github.com/Koenkk/zigbee-herdsman-converters/issues/4769#issuecomment-1436360797)

### What did you expect to happen?

That the switch operated without any delay.

### How to reproduce it (minimal and precise)

Issue presents itself in about 30% of the runs of this scenario

1.  Two Avatto  TS0006 switches are grouped in a HA group (6 entities(gangs) each)
2. Operator requests the Group to Turn ON via switch.turn_on
3. All entities of the switch_1 turns on
4.  5 seconds passes (**ISSUE**)
5.  All entities of switch_2 turn on 

Notes: Also possible that some entities from switch_2 turns on in Step 3, but some of it's entities still wll turn on at Step 4

Extra comments: 

- During the Step 4, The switch responds to physical inputs (touch button) and the entity state correctly reports in HA/Z2M UI (with no delay). But still takes some seconds to the command given in STEP 2. to be enforced. Trying to operate (during step 4) entities individually via Z2M wont work also.

-This also happens if the operator is using switches/entities/gangs individually (e.g. Z2M UI) after some random operations (turning on or off via Z2M) the switch will stop to respond via Z2M. Any command after this moment will be Buffered and the switch will enforce after some seconds (Scenario Step 4) in a short burst (even if it means flickering on and off rapidly)

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle-P USB

### Debug log

.txt
[Zigbee2MQTT:info  2023-02-21 22:06:32: MQTT publis.txt](https://github.com/Koenkk/zigbee2mqtt/files/10799213/Zigbee2MQTT.info.2023-02-21.22.06.32.MQTT.publis.txt)

I'm not sure i got this logs correctly. But the last few commands is the issue from the error scenario.

Philnascimento, 2023-03-21:
You got a solution? mine is not even working with mqtt

rafareusch, 2023-03-21:
No, 

I replaced these Avatto switches with zemismart ones ([link)](https://pt.aliexpress.com/item/1005005131982729.html?spm=a2g0o.detail.0.0.14cf1cf76AVfQ5&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.291025.0&scm_id=1007.13339.291025.0&scm-url=1007.13339.291025.0&pvid=caf96267-f29d-47c2-a9ca-5edd5b8bb4e9&_t=gps-id%3ApcDetailBottomMoreThisSeller%2Cscm-url%3A1007.13339.291025.0%2Cpvid%3Acaf96267-f29d-47c2-a9ca-5edd5b8bb4e9%2Ctpp_buckets%3A668%232846%238108%231977&pdp_ext_f=%7B%22sku_id%22%3A%2212000031788630583%22%2C%22sceneId%22%3A%223339%22%7D&pdp_npi=3%40dis%21BRL%21373.28%21197.5%21%21%21%21%21%40210312ee16794263642036783eb4ac%2112000031788630583%21rec%21BR%21&gatewayAdapt=glo2bra). This zemismart switches are much more stable ( it's recognized as two 3gang switches ).

Sometimes the Zemismart 6 gang still has some small delay (max 1s) to turn on ALL gangs when commanded. But  it's **much** more stable than the Avatto ones.

I've not check if there's any firmware update available with Tuya app.

closed 2023-05-09 by github-actions[bot]