Koenkk/zigbee2mqtt #16777: Sonoff Door Window Sensor

milandzuris, 2023-02-22:
### What happened?

Year and half i have problem only with Sonoff Door Windows Sensor every Sensor after 2-3 months will totaly Disconnected from Z2M, some Sensor need new Battery and some Sensor after repair again working, other device working on 100%, how fix this?

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

 20221102

### Adapter

Sonoff  zStack3x0

meganiekje, 2023-02-23:
I did see similar problems with sonoff door/window switches.
If they are connected directly to the coördinator, they work o.k.
If they are connected via a router, they loose connection after a while.
I replaced the sonoff with those: https://www.zigbee2mqtt.io/devices/TM001-ZA_TM081.html
No more problems.

efolio, 2023-02-25:
I have issues with sonoff sensors as well.

They keep losing connection to Z2M randomly, and the only way to bring them back is to re-pair. Some of them systematically disconnect a few hours after pairing, even though they are right next to the coordinator…

You can see on the door sensors that they still work, and blink when opening/closing. Meaning this may be a connection issue.
When I press the pair button, Z2M notifies that the device leaves the network (even though it had disappeared from the devices list beforehand), before re-discovering it…

I have 20+ of those, so I obviously don’t want to replace them.
Can anybody help us on this ? This is a big issue…

I use the ZB Dongle-E with HAOS and Z2M plugin.

meganiekje, 2023-02-25:
@efolio : I have 10 endpoint devices of different brands and function. When I use the Z2M map function, 8 devices have a permanent green connection to a router or the Sonoff coördinator. Only the 2 sonoff door switches start to 'float' after some days. They have no green connection line. That made me decide to not use them anymore. I did 'rebuild' one of the sonoff door-switches as a licht-switch, by soldering a push-button on it. As long as I use that button at least 2 times a week it keeps working..

milandzuris, 2023-02-25:
@Koenkk

FreddyG88, 2023-03-18:
Is there any possibility to force the devices using directly the coordinator? All door sensors which are close to it work reliable for months now, devices which had to use my TI Router never started working and one which has similar distances to my coordinator and the TI Router stops working after several days. I guess it works as long as it doesn't try to connect via the TI Router.

milandzuris, 2023-03-19:
@Koenkk

efolio, 2023-03-19:
Very strange.
I suspected the shitty battery holders on the door sensors not to make contact properly enough, so I placed a metal washer between the battery and the battery holder, to make it connect very firmly.
I did that on ~60% of the sonoff door sensors (SNZB-04), and since then, no device dropped out of the network !
Even the SNZB-02 temperature sensors that systematically dropped out every day are now stable.

My conclusion on this : one or several devices had a battery issue that made them shit the network, and made some devices on it to drop out.

I still have issues on this network, like some devices pass Unavailable without explanations, and I still have no idea why (even some sector-powered router devices that should be very stable), but there has been no drop out for 2 weeks now !
If anyone has an idea on this, I would be grateful…

Koenkk, 2023-03-19:
It is not possible to force the parent of a zigbee device.

milandzuris, 2023-03-21:
> Very strange. I suspected the shitty battery holders on the door sensors not to make contact properly enough, so I placed a metal washer between the battery and the battery holder, to make it [...]

Please take a Photo of Sensors THX

closed 2023-04-29 by github-actions[bot]

tariqsaidofficial, 2024-09-14:
My Sonoff wifi door sensor stopped working suddenly after 2 weeks and the last battery level was 87. My wifi 2.4Ghz and at  pairing the device detected but does not accept pairing. I tried it via Bluetooth not working. Can anyone help?

[image]

[image]

[image]

[image]