Koenkk/zigbee2mqtt #16778: Pairing only works via Coordinator

m-wo, 2023-02-22:
### What happened?

I´m currently switching my zigbee setup from deconz to z2m. I´ve already integrated a few devices across my house (3 stories), especially the ones close to the coordinator to act as routers for the devices I´m migrating that are more far away from the coordinator.

I´m not trying to add devices that cannot reach the coordinator directly, but I cannot get them paired:
- Debug log shows no sign of the devices (multiple brands/types, from Ikea bulbs to aqara sensors, etc.)
- When I´m moving the device close to the coordinator it works

I had a look at [#7762](https://github.com/Koenkk/zigbee2mqtt/issues/7762) which mentions the same issue but it it should be fixed?
The same way it worked when I originally added all of the devices in deconz.

[image]

### What did you expect to happen?

Pairing of devices via devices acting as routers in the network

### How to reproduce it (minimal and precise)

- Try to pair a device (independent from type/brand, all supported though) not in reach directly to coordinator
- Device won´t pair or even show in the logs, no errors
- Move it close to the coordinator for testing - it will successfully pair

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

6.10.3.0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

nothing shown in the debug log when trying to pair a device

mamrai1, 2023-02-22:
Yes, i do confirm the problem with ezsp frimware 6.10.3.0. Pairing works only via coordinator and near it  when "permit join (all)".  If you want to pair a device via a router you have to select the specific router - permit join(specificrouter).
However after pairing 23 devices i must admit the network is very stable (seems also to better than conbee2)
I have also flashed today the new firmware 7.1.4.0 (ezspv9) on the sonoff zigbee v2 and no problems so far....
i will try to check if pairing problems exist with the newer firmware and report back.

mamrai1, 2023-02-23:
i do confirm problem persists with firmware 7.1.4.0... you have to select the specific router for the pairing.

m-wo, 2023-02-23:
Thanks @mamrai1 - I thought I tested pairing via a specific router already but I tried it today again and it worked to pair the rest of my devices, thanks! At least there’s a workaround…

closed 2023-04-02 by github-actions[bot]

stefanuytterhoeven, 2023-12-04:
Sorry to break in here. I have the same problem with the original Sonoff ZB dongle.
Since a few weeks 1 Hue lamp disconnects from time to time. It reconnects after a few seconds/minutes.
So, I thought: I'll have a look a the map. And, ALL my devices (routers) and end-devices are directly connected to the coordinator (the Sonoff)
so, after some reading, I discovered you could "force" to join via a selected router. So, 2 meters away from that hue lamp, there's another hue lamp (let call it lamp2) . So, I tried the "permit join via lamp2".  first i "removed" lamp1, then I did a "permit join via lamp2". The result: lamp2 was again joined directly to the coordinator. Did I do something wrong?

tenkaibr, 2024-10-07:
I'm having a similar problem. Some switches have duplicate routes and this is causing a problem. I tried to make it connect directly to the cordinator but even when selecting it specifically, it connects to another device. How to force the device to connect to the cordinator?

MajorMaxdom, 2025-11-16:
This still seems to be an issue for me - I cannot pair any devices via any outlet-powered device (which i would describe as a router?).
When putting them closer to the coordinator pairing is no issue.