Koenkk/zigbee2mqtt #16780: Interview fails after deleting and re-pairing the  aqara D1 2 gang smart wall switch QBKG24LM(with neutral wire)

phendrank, 2023-02-22:
### What happened?

i have six aqara D1 2 gang smart wall switch (with neutral wire) QBKG24LM, only 2 device not success to interview.
On first pairing the aqara D1 2 gang QBKG24LM seemed to work but after removing and re-pairing, interview fails and the device shows up as "unsupported".
Potentially similar to this issue, but the dev didn't fix it for me:

#14135
#14288

### What did you expect to happen?

Pair a first time as it did the first and show up in device list as the actual supported device.

### How to reproduce it (minimal and precise)

Pair the QBKG24LM smart switch -- interview completed.

wait a while and re-start the pairing process--device leaves the network then attempts to rejoin -- interview fails.

load up the latest-dev
delete the device (force remove)
plug it in, start the pairing process -- nothing happens, no log entries, no green pop up message about interviewing.

### Zigbee2MQTT version

Current version: 1.30.1-1

### Adapter firmware version

CC2652R_coordinator_20221226

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Debug log

2023-02-22T13:14:13.479Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,14,205,14,205,35]
2023-02-22T13:14:13.495Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2023-02-22T13:14:13.496Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2023-02-22T13:14:13.496Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2023-02-22T13:14:13.496Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2023-02-22T13:14:13.496Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-22T13:14:23.498Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2023-02-22T13:14:23.498Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-02-22T13:14:23.498Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d00078edd31', attempt 6
2023-02-22T13:14:23.498Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2023-02-22T13:14:23.498Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-02-22T13:14:23.500Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d00078edd31' with error 'Error: Interview failed because can not get node descriptor ('0x00158d00078edd31')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:554:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:444:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:567:17)'
2023-02-22T13:14:23.500Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-02-22T13:14:23.513Z zigbee-herdsman:controller:error Interview failed for '0x00158d00078edd31 with error 'Error: Interview failed because can not get node descriptor ('0x00158d00078edd31')'
2023-02-22T13:14:47.015Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,247,169,1,86,73,196]
2023-02-22T13:14:47.015Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,247,169,1,86,73,196]
2023-02-22T13:14:47.015Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [247,169,1,86,73] - 196
2023-02-22T13:14:47.015Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":43511,"relaycount":1,"relaylist":[18774]}
2023-02-22T13:14:47.016Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-22T13:14:58.522Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,236,188,2,3,58,14,205,46,254,25,68,129,0,0,10,0,236,188,1,1,0,94,0,203,188,52,0,0]
2023-02-22T13:14:58.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,236,188,2,3,58,14,205,46,254,25,68,129,0,0,10,0,236,188,1,1,0,94,0,203,188,52,0,0]
2023-02-22T13:14:58.522Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [236,188,2,3,58,14,205] - 46
2023-02-22T13:14:58.522Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":48364,"relaycount":2,"relaylist":[14851,52494]}
2023-02-22T13:14:58.523Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,10,0,236,188,1,1,0,94,0,203,188,52,0,0]
2023-02-22T13:14:58.525Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [5,16,130,0,0,0,14,205,27,212]
2023-02-22T13:14:58.525Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,10,0,236,188,1,1,0,94,0,203,188,52,0,0,5,16,130,0,0,0,14,205,27,212]
2023-02-22T13:14:58.525Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,10,0,236,188,1,1,0,94,0,203,188,52,0,0,5,16,130,0,0,0,14,205,27] - 212
2023-02-22T13:14:58.525Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":10,"srcaddr":48364,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":3456203,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[16,130,0,0,0]}}
2023-02-22T13:14:58.527Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":130,"manufacturerCode":null,"commandIdentifier":0},"Payload":[{"attrId":0}],"Command":{"ID":0,"name":"read","parameters":[{"name":"attrId","type":33}],"response":1}},"address":48364,"endpoint":1,"linkquality":94,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-02-22T13:14:58.538Z zigbee-herdsman:controller:endpoint ReadResponse 0x00158d000398d653/1 genTime({"time":730386899}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[... 11 lines cut ...]
2023-02-22T13:14:58.563Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,95] - 153
2023-02-22T13:14:58.563Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":95}
2023-02-22T13:14:58.564Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-22T13:15:08.823Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2023-02-22T13:15:08.824Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2023-02-22T13:15:08.843Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3,69,182,0,0,0,240]
2023-02-22T13:15:08.843Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3,69,182,0,0,0,240]
2023-02-22T13:15:08.843Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2023-02-22T13:15:08.843Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2023-02-22T13:15:08.844Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2023-02-22T13:15:08.844Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2023-02-22T13:15:08.844Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2023-02-22T13:15:08.844Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-02-22T13:15:08.845Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":96,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,98,2,11,254,0]}}
2023-02-22T13:15:08.845Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,96,0,30,6,0,25,98,2,11,254,0,233]
2023-02-22T13:15:08.853Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2023-02-22T13:15:08.853Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2023-02-22T13:15:08.853Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2023-02-22T13:15:08.853Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
[... cut ...]
2023-02-22T13:15:08.940Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,96,85]
2023-02-22T13:15:08.940Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,96,85]
2023-02-22T13:15:08.940Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,96] - 85
2023-02-22T13:15:08.941Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":96}
2023-02-22T13:15:08.941Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Spica2000, 2023-03-11:
I have a similar problem, but for me initial pairing is also failed, interview is not completed.

pplu19, 2023-03-16:
I am also having the same issues with a QBKG24LM. I have a SONOFF Zigbee 3.0 adapter with 20220219 firmware version.

Spica2000, 2023-04-16:
Still no news? The problem is still exists.

phendrank, 2023-05-01:
yes. still exists.

closed 2023-06-09 by github-actions[bot]

lbialy, 2023-07-20:
Problem showed up for during migration from Conbee2 to CC2652RB Slae.sh stick, both QBKG24LM and QBKG23LM fail to pair again. I'm using Synology DS720+ with dockerized Z2M with mosquitto. I had to add serial USB drivers from [this Jadahl.com dump](https://github.com/robertklep/dsm7-usb-serial-drivers). I have flashed Slae.sh stick to fix the issue with `CC2652RB_coordinator_20230507` firmware, didn't help. Z2M was cleaned completely with all files removed several times. After that I have aborted the migration to Slae.sh, rmmoded usb drivers, restored original configuration (but not the database and state.json files) for Conbee2 and restarted Z2M only to find out that D1 gang wall switches that have attempted to be paired with CC2652RB stick can't join Conbee2 based network with the same problem (stuck at interview, then fail to pair). Most of the other switches joined the Conbee2 network successfully again after I killed the power to them for several minutes (some are still borked though) but the ones that tried to pair with slae.sh can't rejoin. 

Here's the log from herdsman, basically the same thing as OP:
```
Zigbee2MQTT:info  2023-07-20 15:44:15: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:info  2023-07-20 15:44:15: Connected to MQTT server
2023-07-20T13:44:15.639Z zigbee-herdsman:controller:database:log Writing database to '/app/data/database.db'
Zigbee2MQTT:info  2023-07-20 15:44:15: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-07-20 15:44:15: MQTT publish: topic 'zigbee2mqtt/Światło Korytarz/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-07-20 15:44:15: Started frontend on port 0.0.0.0:8834
Zigbee2MQTT:info  2023-07-20 15:44:15: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-07-20 15:44:15: Zigbee2MQTT started!
2023-07-20T13:45:59.502Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,208,68,163,63,242,6,0,141,21,0,0,0,231]
2023-07-20T13:45:59.502Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,208,68,163,63,242,6,0,141,21,0,0,0,231]
2023-07-20T13:45:59.502Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [208,68,163,63,242,6,0,141,21,0,0,0] - 231
2023-07-20T13:45:59.503Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":17616,"extaddr":"0x00158d0006f23fa3","parentaddr":0}
2023-07-20T13:45:59.503Z zigbee-herdsman:controller:log Device '0x00158d0006f23fa3' joined
2023-07-20T13:45:59.503Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-07-20T13:45:59.503Z zigbee-herdsman:controller:log Device '0x00158d0006f23fa3' accepted by handler
2023-07-20T13:45:59.503Z zigbee-herdsman:controller:log New device '0x00158d0006f23fa3' joined
2023-07-20T13:45:59.504Z zigbee-herdsman:controller:log Creating device '0x00158d0006f23fa3'
2023-07-20T13:45:59.504Z zigbee-herdsman:controller:database:log Writing database to '/app/data/database.db'
Zigbee2MQTT:info  2023-07-20 15:45:59: Device 'Włącznik - Toaleta' joined
Zigbee2MQTT:info  2023-07-20 15:45:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Włącznik - Toaleta","ieee_address":"0x00158d0006f23fa3"},"type":"device_joined"}'
2023-07-20T13:45:59.591Z zigbee-herdsman:controller:log Interview '0x00158d0006f23fa3' start
Zigbee2MQTT:info  2023-07-20 15:45:59: Starting interview of 'Włącznik - Toaleta'
Zigbee2MQTT:info  2023-07-20 15:45:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Włącznik - Toaleta","ieee_address":"0x00158d0006f23fa3","status":"started"},"type":"device_interview"}'
2023-07-20T13:45:59.597Z zigbee-herdsman:controller:device:log Interview - start device '0x00158d0006f23fa3'
2023-07-20T13:45:59.598Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":17616,"nwkaddrofinterest":17616}
2023-07-20T13:45:59.598Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,208,68,208,68,35]
2023-07-20T13:45:59.599Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,208,68,208,68,163,63,242,6,0,141,21,0,142,247]
2023-07-20T13:45:59.599Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,208,68,208,68,163,63,242,6,0,141,21,0,142,247]
2023-07-20T13:45:59.599Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [208,68,208,68,163,63,242,6,0,141,21,0,142] - 247
2023-07-20T13:45:59.600Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":17616,"nwkaddr":17616,"ieeeaddr":"0x00158d0006f23fa3","capabilities":142}
2023-07-20T13:45:59.600Z zigbee-herdsman:controller:log Device announce '0x00158d0006f23fa3'
Zigbee2MQTT:info  2023-07-20 15:45:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Włącznik - Toaleta","ieee_address":"0x00158d0006f23fa3"},"type":"device_announce"}'
2023-07-20T13:45:59.605Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-07-20T13:45:59.607Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2023-07-20T13:45:59.607Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2023-07-20T13:45:59.607Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2023-07-20T13:45:59.607Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
[... cut ...]
2023-07-20T13:47:44.739Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2023-07-20T13:47:44.739Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-07-20T13:47:54.743Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2023-07-20T13:47:54.743Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-07-20T13:47:54.743Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0006f23fa3', attempt 5
```

baljinder1082, 2024-08-13:
same issue here, maybe switch is faulty?

lbialy, 2024-08-14:
This was actually solved here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/412

tl;dr: aqara ota update borks the switch (some might come with faulty firmware), the only way to fix is to dismantle and reflash the switch with original firmware.