Koenkk/zigbee2mqtt #16782: Upgrade issue - npm ci - error Path is a directory

Tibronx, 2023-02-22:
### What happened?

I tried to upgrade on January 25th - I got this error and didn't handle it.
```
pi@raspberrypi:/opt/zigbee2mqtt $ sudo cat  /root/.npm/_logs/2023-01-25T16_57_14_026Z-debug-0.log
[...]
42 timing command:ci Completed in 3826ms
43 verbose stack SystemError [ERR_FS_EISDIR]: Path is a directory: rm returned EISDIR (is a directory) /opt/zigbee2mqtt/node_modules/.bin
43 verbose stack     at new SystemError (internal/errors.js:217:5)
43 verbose stack     at new NodeError (internal/errors.js:302:7)
43 verbose stack     at internal/fs/utils.js:718:23
43 verbose stack     at callback (/usr/lib/node_modules/npm/node_modules/graceful-fs/polyfills.js:306:20)
43 verbose stack     at FSReqCallback.oncomplete (fs.js:193:5)
44 verbose cwd /opt/zigbee2mqtt
45 verbose Linux 4.19.66-v7+
46 verbose node v14.20.0
47 verbose npm  v9.3.0
48 error code ERR_FS_EISDIR
49 error syscall rm
50 error path /opt/zigbee2mqtt/node_modules/.bin
51 error errno 21
52 error Path is a directory: rm returned EISDIR (is a directory) /opt/zigbee2mqtt/node_modules/.bin
53 verbose exit 21
54 timing npm Completed in 4128ms
55 verbose code 21
56 error A complete log of this run can be found in:
56 error     /root/.npm/_logs/2023-01-25T16_57_14_026Z-debug-0.log
```

Today I tried again when I found the courage to do it and same error again.
Even if on frontend it's stating that I'm on latest version, It's totally inconsistent as the upgrade process never finished properly in a month, having same error continously.

### What did you expect to happen?

Upgrade with no error
npm ci executing properly

### How to reproduce it (minimal and precise)

Try to upgrade

### Zigbee2MQTT version

It's saying 1.30.1 but last upgrades failed...I might have configuration inconsistencies

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

pi@raspberrypi:/opt/zigbee2mqtt $ sudo ./update.sh
Stopping Zigbee2MQTT...
Creating backup of configuration...
Updating...
Updating 14021395..eb878d3d
Checking out files: 100% (13/13), done.
Fast-forward
 .github/workflows/ci.yml       |    4 +-
 docker/Dockerfile              |    2 +-
 lib/controller.ts              |    5 +-
 lib/extension/bind.ts          |    3 +
 lib/extension/frontend.ts      |    9 +-
 lib/extension/homeassistant.ts |   16 ++-
 lib/extension/otaUpdate.ts     |    8 +-
 lib/util/settings.schema.json  |    4 +-
 package-lock.json              | 1769 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----------------------------------------------------------------------------------------------------------------------------------------------------
 package.json                   |   36 +++----
 test/bridge.test.js            |    4 +-
 test/controller.test.js        |    1 -
 test/otaUpdate.test.js         |    6 +-
 13 files changed, 967 insertions(+), 900 deletions(-)
Installing dependencies...
npm ERR! code ERR_FS_EISDIR
npm ERR! syscall rm
npm ERR! path /opt/zigbee2mqtt/node_modules/.bin
npm ERR! errno 21
npm ERR! Path is a directory: rm returned EISDIR (is a directory) /opt/zigbee2mqtt/node_modules/.bin

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2023-02-22T14_54_16_889Z-debug-0.log

pi@raspberrypi:/opt/zigbee2mqtt $ sudo more /root/.npm/_logs/2023-02-22T14_54_16_889Z-debug-0.log
0 verbose cli /usr/bin/node /usr/bin/npm
1 info using npm@9.3.0
2 info using node@v14.20.0
3 timing npm:load:whichnode Completed in 4ms
4 timing config:load:defaults Completed in 13ms
5 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 17ms
6 timing config:load:builtin Completed in 18ms
7 timing config:load:cli Completed in 22ms
8 timing config:load:env Completed in 2ms
9 timing config:load:file:/opt/zigbee2mqtt/.npmrc Completed in 5ms
10 timing config:load:project Completed in 18ms
11 timing config:load:file:/root/.npmrc Completed in 0ms
12 timing config:load:user Completed in 2ms
13 timing config:load:file:/usr/etc/npmrc Completed in 1ms
14 timing config:load:global Completed in 1ms
15 timing config:load:setEnvs Completed in 4ms
16 timing config:load Completed in 82ms
17 timing npm:load:configload Completed in 83ms
18 timing npm:load:mkdirpcache Completed in 1ms
19 timing npm:load:mkdirplogs Completed in 2ms
20 verbose title npm ci
21 verbose argv "ci"
22 timing npm:load:setTitle Completed in 7ms
23 timing config:load:flatten Completed in 15ms
24 timing npm:load:display Completed in 20ms
25 verbose logfile logs-max:10 dir:/root/.npm/_logs/2023-02-22T14_54_16_889Z-
26 verbose logfile /root/.npm/_logs/2023-02-22T14_54_16_889Z-debug-0.log
27 timing npm:load:logFile Completed in 23ms
28 timing npm:load:timers Completed in 1ms
29 timing npm:load:configScope Completed in 0ms
30 timing npm:load Completed in 143ms
31 timing arborist:ctor Completed in 4ms
32 silly logfile start cleaning logs, removing 1 files
33 silly logfile done cleaning log files
34 timing idealTree:init Completed in 28ms
35 timing idealTree:userRequests Completed in 1ms
36 silly idealTree buildDeps
37 timing idealTree:#root Completed in 2ms
38 timing idealTree:buildDeps Completed in 8ms
39 timing idealTree:fixDepFlags Completed in 0ms
40 timing idealTree Completed in 162ms
41 timing npm-ci:rm Completed in 38ms
42 timing command:ci Completed in 3928ms
43 verbose stack SystemError [ERR_FS_EISDIR]: Path is a directory: rm returned EISDIR (is a directory) /opt/zigbee2mqtt/node_modules/.bin
43 verbose stack     at new SystemError (internal/errors.js:217:5)
43 verbose stack     at new NodeError (internal/errors.js:302:7)
43 verbose stack     at internal/fs/utils.js:718:23
43 verbose stack     at callback (/usr/lib/node_modules/npm/node_modules/graceful-fs/polyfills.js:306:20)
43 verbose stack     at FSReqCallback.oncomplete (fs.js:193:5)
44 verbose cwd /opt/zigbee2mqtt
45 verbose Linux 4.19.66-v7+
46 verbose node v14.20.0
47 verbose npm  v9.3.0
48 error code ERR_FS_EISDIR
49 error syscall rm
50 error path /opt/zigbee2mqtt/node_modules/.bin
51 error errno 21
52 error Path is a directory: rm returned EISDIR (is a directory) /opt/zigbee2mqtt/node_modules/.bin
53 verbose exit 21
54 timing npm Completed in 4197ms
55 verbose code 21
56 error A complete log of this run can be found in:
56 error     /root/.npm/_logs/2023-02-22T14_54_16_889Z-debug-0.log

lux73, 2023-02-23:
try upgrade nodeJS to v18.x

after that do a `npm rebuild`

Tibronx, 2023-02-23:
Just tried to do it but not successfull upgrade - don't know why but my raspbian os is not detected
```
pi@raspberrypi:~ $ sudo curl -fsSL https://deb.nodesource.com/setup_18.x | sudo -E bash -

## Installing the NodeSource Node.js 18.x repo...

## Populating apt-get cache...

+ apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:3 https://deb.nodesource.com/node_14.x stretch InRelease
Fetched 15.0 kB in 1s (11.7 kB/s)
Reading package lists... Done
Traceback (most recent call last):
  File "/usr/bin/lsb_release", line 25, in <module>
    import lsb_release
ModuleNotFoundError: No module named 'lsb_release'
Traceback (most recent call last):
  File "/usr/bin/lsb_release", line 25, in <module>
    import lsb_release
ModuleNotFoundError: No module named 'lsb_release'

## Confirming "" is supported...

+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_18.x/dists//Release'

## Your distribution, identified as "", is not currently supported, please contact NodeSource at https://github.com/nodesource/distributions/issues if you think this is incorrect or would like your distribution to be considered for support
```

Looking at the script, seems that I was missing lsb_release command so I installed lsb_core via apt but get same result:
```
pi@raspberrypi:~ $ sudo lsb_release -c -s
Traceback (most recent call last):
  File "/usr/bin/lsb_release", line 25, in <module>
    import lsb_release
ModuleNotFoundError: No module named 'lsb_release'
```

Here are my raspian info (old one i know):
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@raspberrypi:~ $ uname --all
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

lux73, 2023-02-23:
I would assume stretch isn't supported anymore

must update to at least buster (10.x) oder better bullseye (11.x)

Without there is no luck

closed 2023-04-02 by github-actions[bot]